Klipper3d/klipper #3289: Cannot get DuetWifi+Duex5 to connect reliably to a Ubuntu 20.04 LTS PC

mdealer, 2020-09-06:
Due to a freak coincidence I lost all my Klipper data (back ups of the printer configuration etc), so, sorry, no log file here.

The symptoms were rather consistent. The connection to the DuetWifi was extremely flaky. One in 20 tries would actually connect and appear to work. If the connection somehow got interrupted (by a restart for example) it would plain refuse to connect. Tried different cables, different USB ports, reflashing, rebuilding etc. I even tried another PC, but this time on a Windows Host and Linux VM using Virtual Box USB port forwarding. Everything could be forwarded no problem, UP UNTIL the actual Klipper firmware would start. From this point, that's it, the device appears as "busy" forever, unable to forward. If I press ERASE and RESET on the board, then SAMBA port opens completely normal. If I flash the Duet firmware, then it also acts normal and the port can be forwarded to VM without problems. I am flashing etc entirely from within the VM.

I get the feeling the problem is somehow the firmware and wanted to ask before I keep trying this even more PCs.

Is there anyone being able to use latest master branch Klipper with DuetWifi without RPi around here? Any tips would be appreciated.

closed 2020-09-13 by klipper-gitissuebot

mdealer, 2020-09-13:
> 
> 
> It appears the Klipper log file was not provided.
> I'm closing this now, sorry. Feel free to attach the log to this ticket sometime in the future, or create a new issue once you can provide the log file.

But what is so special about the way Klipper handles USB here that it won't work with Windows host + Linux guest in VirtualBox? That's the main question here. There will never be any log for this issue this way.

BlackStump, 2020-09-14:
[Please read this section - Can I run Klipper on something other than a Raspberry Pi 3](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3)
Klipper works as intended on supported hardware!