Klipper3d/klipper #3290: Is there a way to disable temperature checks?

cfelicio, 2020-09-06:
When changing the nozzle, for example, klipper will notice the sudden drop in temperature on the extruder (due to me touching it with a tool, the temp drops).

I did not notice this at first, and kept unscrewing the nozzle, eventually noticed the nozzle was hard to unscrew, and that's when I saw Klipper had turned off the heater. How do I make sure the heater stays on in such unique / temporary situations?

KevinOConnor, 2020-09-08:
The only way to disable the temperature checks is to provide a very large range to min_temp/max_temp, and/or to provide a very large value to the max_error setting of verify_heater.

-Kevin

cfelicio, 2020-09-08:
Thanks Kevin! Will try that!

closed 2020-09-08 by cfelicio