Klipper3d/klipper #3292: alfawise U30 LCD Support?

linux-paul, 2020-09-07:
Is possible get an alfawise U30 LCD Display running? Unfortunately I can't find a configuration example using this connection type. This may the pinmap within Marlin:
/**
 * Note: Alfawise screens use various TFT controllers. Supported screens
 * are based on the ILI9341, ILI9328 and ST7798V. Define init sequences for
 * other screens in u8g_dev_tft_320x240_upscale_from_128x64.cpp
 *
 * If the screen stays white, disable 'LCD_RESET_PIN' to let the bootloader
 * init the screen.
 *
 * Setting an 'LCD_RESET_PIN' may cause a flicker when entering the LCD menu
 * because Marlin uses the reset as a failsafe to revive a glitchy LCD.
 */

#define LCD_RESET_PIN                       PC4   // pin 33
#define LCD_BACKLIGHT_PIN                   PD12  // pin 59
#define FSMC_CS_PIN                         PD7   // pin 88 = FSMC_NE1
#define FSMC_RS_PIN                         PD11  // pin 58 A16 Register. Only one address needed

#define LCD_USE_DMA_FSMC                          // Use DMA transfers to send data to the TFT
#define FSMC_DMA_DEV                        DMA2
#define FSMC_DMA_CHANNEL                 DMA_CH5

#define DOGLCD_MOSI                         -1    // Prevent auto-define by Conditionals_post.h
#define DOGLCD_SCK                          -1
[klippy.log](https://github.com/KevinOConnor/klipper/files/5183550/klippy.log)

KevinOConnor, 2020-09-11:
As far as I know, that display is not supported.  In order for it to be supported, an interested developer would need to take it on and submit the work.

-Kevin

linux-paul, 2020-09-11:
Thank you, Request closed so far.

closed 2020-09-11 by linux-paul

elegos, 2020-11-17:
@KevinOConnor if I had free time (I wish I had!!), where should I start to work on it? I've got this printer and I'm a former software engineer, even though I should dust off my C knowledge and I have no prior screen experience.

Thanks :100:

KevinOConnor, 2020-11-18:
Alas, I don't know enough about that display to even give any hints on where to start.

-Kevin