Klipper3d/klipper #3293: Set current hight as Z=0 and save

robvoi, 2020-09-07:
Not sure if I can ask usage questions here but I didn't find a forum or such.

Is there a way to set the current hight as Z=0 from GCode/Terminal?
Or to define an offset which also applies after a reboot.

I use the Coordinate Offset (via the Klipper Plugin for Octoprint) but this doesn't safe the offset permanently.

What I want to achieve: with two fixed Z-Endstops (hall effect) I stop both Z-axis at a specific height. This levels the Z axis.
Now I am 0.X mm off the bed. This I easily want to set via console by moving 0.05 up or down until Z is perfect. Then save.

schnello, 2020-09-10:
>Is there a way to set the current hight as Z=0 from GCode/Terminal?

I use SET_KINEMATIC_POSITION Z=0

robvoi, 2020-09-11:
> I use SET_KINEMATIC_POSITION Z=0
Thanks. But this doesn't save until after a reset, does it?
That's what I want I need.

schnello, 2020-09-11:
If you want apply an offset use the SET_GCODE_OFFSET command

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#extended-g-code-commands

robvoi, 2020-09-11:
Again thanks.
But ... Also this doesn't survive a restart.

KevinOConnor, 2020-09-11:
If you have a hall effect endstop, then your can set that as the Z endstop and use the position_endstop field to set the position.  Otherwise, there is no ability to save an offset in the config file.  However, if one updates their slicer to run a macro at each print start, then one can place a preset offset in that macro.

-Kevin

robvoi, 2020-09-11:
> If you have a hall effect endstop, then your can set that as the Z endstop and use the position_endstop field to set the position. Otherwise, there is no ability to save an offset in the config [...]

Hi Kevin,
first .... great firmware. Really great.

Thanks a lot for your advise.
position_endstop works - "but" I have to do it at two spots. (z0 & z1)
I now solved it with a macro that sets the offset. The slices calls the macro.
Only one position in the config. And the sliced modes work with changing offsets.

What does't work is to do all this from OctoPod (iPhone) as I can't change the config from there.
But that's a pure comfort thing.

Maybe something for the future? Set and store Z offset from Terminal OR save current position to config as Z=0.

closed 2020-10-11 by github-actions[bot]