Klipper3d/klipper #3294: Config error

epsaty, 2020-09-07:
Hi everyone, 

Updated Klipper to latest version from an older one and it seems that some parameters in the config file are not being accepted anymore, like syntax has been changed in the new version.

Error:
Internal error during connect: Choice 'item' for option 'type' is not a valid choice

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
Internal error during connect: Choice 'item' for option 'type' is not a valid choice

[klippy.log](https://github.com/KevinOConnor/klipper/files/5184422/klippy.log)

Any help is apreciated.

Thank you*

theopensourcerer, 2020-09-07:
Probably documented in here (especially this part: `20200809: The menu system has been rewritten. If the menu has been customized then it will be necessary to update to the new configuration. See config/example-menu.cfg for configuration details and see klippy/extras/display/menu.cfg for examples.`):

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md

epsaty, 2020-09-07:
Thanks for the guidance, managed to get it working somehow.

For anyone how was using this config found o various tutorials/resources... using the new version it looks something like this:

[bed_mesh]
mesh_min: 30,30
mesh_max: 210,210
probe_count: 4,4
fade_start: 1.0
fade_end: 10.0
speed: 100
algorithm: lagrange

[menu __main __setup __calib __meshBed]
type: list
name: Mesh Bed Level
	
[menu __main __setup __calib __meshBed __start]
type: command
name: Start probing
gcode: 
	G28
	BED_MESH_CALIBRATE
	TESTZ Z=-5

[menu __main __setup __calib __meshBed __zposition]
type: input
name: Z: {'%05.3f' % menu.input}
input: {printer.gcode_move.gcode_position.z}
input_step: 1
realtime: True

[menu __main __setup __calib __meshBed __zNudgeDown]
type: command
name: Nudge Down
gcode: TESTZ Z=-0.025

[menu __main __setup __calib __meshBed __zNudgeUp]
type: command
name: Nudge Up
gcode: TESTZ Z=0.025

[menu __main __setup __calib __meshBed __next]
type: command
name: Next
gcode: 
	ACCEPT
	TESTZ Z=-5
	
[menu __main __setup __calib __meshBed __save]
type: command
name: Save & Restart
gcode: SAVE_CONFIG

[menu __main __setup __calib __bedScrews]
type: list
name: Adjust screws

[menu __main __setup __calib __bedScrews __start]
type: command
name: Start
gcode: G28
       BED_SCREWS_ADJUST

[menu __main __setup __calib __bedScrews __accept]
type: command
name: Accept
gcode: ACCEPT

[menu __main __setup __calib __bedScrews __adjusted]
type: command
name: Adjusted
gcode: ADJUSTED

[menu __main __setup __calib __bedScrews __save]
type: command
name: Save config
gcode: SAVE_CONFIG

[menu __main __setup __calib __bedScrews __abort]
type: command
name: Abort
gcode: ABORT

closed 2020-09-07 by epsaty