Klipper3d/klipper #3295: Cannot install Klipper

Drows, 2020-09-08:
Hello all, i know this isn't an issue with klipper itself but i've been getting this error and idk what else to do
this is what shows on my console:

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Also creating executable in /home/pi/klippy-env/bin/python
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting setuptools
  Using cached https://files.pythonhosted.org/packages/e1/b7/182161210a13158cd3ccc41ee19aadef54496b74f2817cc147006ec932b4/setuptools-44.1.1-py2.py3-none-any.whl
Collecting pkg_resources
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pkg-resources/

----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel failed with error code 1

so apparently the website that its trying to use doesnt exist anymore...

Drows, 2020-09-08:
I managed to install pip but now I get this error:
./klipper/scripts/install-octopi.sh: line 40: /home/pi/klippy-env/bin/pip: No such file or directory
Still no log file, trying to get this to work

cfelicio, 2020-09-08:
Same here, likely related to something similar to this: https://github.com/KevinOConnor/klipper/issues/2383

cfelicio, 2020-09-08:
I wonder if upgrading to Python 3 can solve the problem. Going to try this for now:

https://docs.octoprint.org/en/master/plugins/python3_migration.html

jaayyemmm, 2020-09-08:
Same issue for me, were you able to resolve it? It seems that the packages related to the link below doesnt exist anymore. I tried accessing the site and its a 404 error.

https://www.piwheels.org/simple/pkg-resources/

theopensourcerer, 2020-09-08:
Klipper currently works on Python 2 not Python 3. There is a branch for Python 3 support but it is not fully tested/supported or merged into the master branch.

I just installed Klipper the other day on a clean Rasbian and it worked fine. Are you using the "master" branch or are you using the 0.8.0 release? If the later - I would recommend switching to the master branch. It looks like there was a fix for this issue merged in January:

https://github.com/KevinOConnor/klipper/commit/5fc92ff6b7f2c824346fdfd559f2d44978f380ab

RentableSocks, 2020-09-08:
I am also running into this issue because of the packages being relocated or removed.

@theopensourcerer I had just installed klipper main branch this morning at ~5AM PST and when I tried to install again this afternoon (~3-5PM PST) this behavior started, so I suspect the change happened somewhere between those two times.

Drows, 2020-09-08:
I still wasn't able to make it work, despite trying with a fresh OS install

Drows, 2020-09-08:
> I wonder if upgrading to Python 3 can solve the problem. Going to try this for now:
> 
> https://docs.octoprint.org/en/master/plugins/python3_migration.html

did you manage to make it work? i tried using python 3 but wasn't able to fix the issue

adamnock, 2020-09-08:
Same issue here, fresh install of klipper on a OctoPi which has both py3 and py2.7

It shouldnt be conflict with py3 as Klipper's installed explicitely references py2.7 and sym links it, then tries to instance a venv for itself.
Its failing collecting pkg-resources from piwheels, which following the link manually does appear to be missing

Possibly an upstream dependency been removed?

adamnock, 2020-09-08:
Suspicion it might be an issue with the version of virtualenv installed onto Rasbian Buster (15.1).
If I remove that, install the pip version (20+) and rerun the failing command manually, its successful.

Investigating if the Rasbian package has an update in another repo.

adamnock, 2020-09-08:
Hmm interesting

OK, workaround

`sudo apt remove virtualenv python-virtualenv python3-virtualenv`
`sudo pip uninstall virtualenv` (Dont worry if this fails)
`sudo pip install virtualenv`
`virtualenv --download -p python2 ~/klippy-env/`

Then rerun the Klipper installer. Itll reinstall the apt venv packages but because the venv exists it wont fail out this time

Still going to find a root cause though

Drows, 2020-09-08:
I'm trying your workaround right now, will update if it works

Drows, 2020-09-08:
I now get this error but I don't think it's related.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00

adamnock, 2020-09-08:
Nope, thats an issue with whatever you are doing to flash the printer board. Look up help specific to the printer or mainboard you have and flashing using avrdude or flashing klipper specifically.

jaayyemmm, 2020-09-08:
> Hmm interesting
>
> OK, workaround
>
> `sudo apt remove virtualenv python-virtualenv python3-virtualenv`
> `sudo pip uninstall virtualenv` (Dont worry if this fails)
> `sudo pip install virtualenv` [...]

Also worked for me, thank you!

RentableSocks, 2020-09-08:
Looks like things are normal now, installed successfully

cfelicio, 2020-09-08:
Same here, things seem to be working again!

Coffee0297, 2020-09-12:
did not work for me on ubuntu server 20.04.. we really need to get to python3 bc more and more distros drop support for it. it makes installing it a pain.

closed 2020-09-15 by klipper-gitissuebot