Klipper3d/klipper #3297: MCU 'z' shutdown: No next step on Force_move

Stephan3, 2020-09-08:
Hello Kevin.

One of my printers had that every now and then when i used FORCE_MOVE command. Since i am not using DWC on klippy thread anymore(thanks for the unixsocket merge btw), i can be sure thats not related to it. 

I am able to run FORCE_MOVE on the other 3 Vorons without an issue. The one thats failing differs only in stepper drivers. 3 printers have TMC2209 and the one thats failing has TMC2208. Once i run Forcemove with speed 2-5mm/s it works in 8 of 10 trys. MOVE_SPEED=25 is failing reliably.

Short log is attached, focusing on that.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5187285/klippy.log)

Thanks
Stephan

KevinOConnor, 2020-09-08:
> One of my printers had that every now and then when i used FORCE_MOVE command.

The `max_accel` value in the config is indirectly programmed into the micro-controller as a sanity check.  If a FORCE_MOVE is made to a kinematic stepper without acceleration then it can lead to these "No next step" errors.  Either specify an ACCEL (less than or equal to max_accel) or use a very low VELOCITY.  I've updated the docs to note this (commit 91979756).

> Since i am not using DWC on klippy thread anymore(thanks for the unixsocket merge btw),

Thanks.  Will the main "dwc2 for klipper" be updated to use the "webhooks unix socket" as well?  It would be great to be able to offer Klipper support to those users that wish to run dwc2.

-Kevin

Stephan3, 2020-09-08:
> The max_accel value in the config is indirectly programmed into the micro-controller as a sanity check. If a FORCE_MOVE is made to a kinematic stepper without acceleration then it can lead to these [...]

I will try that and close this ticket for now. Its strange that only the tmc2208 equiped rig is doing this.

> Thanks. Will the main "dwc2 for klipper" be updated to use the "webhooks unix socket" as well? It would be great to be able to offer Klipper support to those users that wish to run dwc2.

As i am a hobiest, there are so many "things" in the existing version. So i decided to rewrite the whole thing and give every solution a review. The current version is live on all my 4 rigs and gets improvements during my day to day print jobs. Its in a usable state and runs standalone. @Arksine helped a lot to make me understnading the details od the socket API. MAkes my life so much easier.

https://github.com/Stephan3/dwc2-for-klipper-socket

no documentation no nothing. you put the webinterface to ./web/ and run it using python3. it will came up and run. No crashes so far but still potential in covering all functionality.

closed 2020-09-08 by Stephan3