Klipper3d/klipper #3298: FR: Sensorless homing for TMC2660

pandaym3D, 2020-09-08:
Hi!

I am running Klipper on my Prusa Bear, and would like to expand on some functionality. I have therefore invested in a Duet WiFi for it, and see that the TMC2660 are supported. I however understand that sensorless homing has not been implemented for these drivers.
Any chance this could be added?

Thanks for a great project!

Dummy klippy:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5188046/klippy.log)

inks007, 2020-09-08:
See [https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg)

trevjonez, 2020-09-08:
afaik the only way to get a sensorless virtual endstop on a TMC driver is via [TMCVirtualPinHelper](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/tmc.py#L184)
and if we search for a reference to the pin helper in [tmc2660.py](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/tmc2660.py) we come up empty 😢 

a quick skim of the [2660 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2660_datasheet.pdf) has the stallguard out on the `SG_TST` pin which I can't find referenced anywhere in klippy code or example-extras.

additionally the stallguard can be read via `DRVSTATUS` register in all three configs of `RDSEL`, though I would expect using that register as an endstop would require some rework in c code? so if SG_TST is hooked to an MCU pin implementing against that would be far simpler. unless there is already some sort of spi endstop support?

skimming [the duet2 wifi schematic](https://github.com/T3P3/Duet/blob/master/Duet2/Duet2v1.04/DuetWifi_Schematic_v1.04.pdf) it looks like each driver's SG_TST is connected to an input pin of a 74HC32 which ties back to a single pin (PB5) on the SAM4E8E mcu.
so all drivers that are on the main duet board share a single pin for their endstop. which may be a limitation on when you can or can't sensorless home depending on kinematic setup? maybe that would make using the SPI register reading preferable if latency were not an issue? at least on something like a delta or corexy. all of that is just limitations of that one particular PCB's implementation though.

long story short: klipper's current tmc sensorless homing code requires the presence of a `diag*` pin, which is not supported by the current tmc2660 code, and the hardware of the tmc2660 doesn't use a general purpose diag pin the way the 2209, 2130, and 5160 do. The 2660 uses a specific active high sg_tst pin.

inks007, 2020-09-09:
The examples do show that you can configure `driver_SGT`. You're right about there not being a configurable diag pin, but a dedicated stall SG_TST pin. These are all OR'ed using the 74HC32 to a single MCU pin.

Since Klipper only does endstop checks whilst homing, only one stepper would stall (two for coreXY, but moot point) . You could just specify `endstop_pin: PB5` for ONE stepper section and it would probably work.

I'm not sure you can specify the same input pin in different configuration sections and I don't think Klipper's `multi_pin` function works for duplicating input pins (though I haven't tested).

KevinOConnor, 2020-09-11:
FYI, I don't know about the particular capabilities of the tmc2660 driver.  If it supports sensorless homing, then I suspect it could be implemented in Klipper.  If so, an interested developer would need to take that on and submit it.

-Kevin

closed 2020-10-11 by github-actions[bot]

RedGuilhas, 2021-02-14:
Is there any way of getting sensorless homing to work on this drivers?
Thanks

trevjonez, 2021-02-14:
> Is there any way of getting sensorless homing to work on this drivers?
> Thanks

the 2660 extra needs support added to it for it to work. Just needs someone to pick up the work.

KevinOConnor, 2021-02-27:
FYI, looking at the tmc2660 spec, there does not appear to be anything special that needs to be done for sensorless homing.  One should be able to set the axis endstop_pin to the pin wired to the SG_TST pin on the tmc2660, and configure the driver_SGT value according to the description in docs/Sensorless_Homing.md .  As far as I can tell, it should work.

-Kevin

frankjakobsen, 2021-03-14:
> FYI, looking at the tmc2660 spec, there does not appear to be anything special that needs to be done for sensorless homing. One should be able to set the axis endstop_pin to the pin wired to the [...]

On the Duet cards it seems to be wired from the tmc2660 to an or-gate (74hc32) and then to a common PB5 input in the MCU. Would it be enough to specify PB5 for X or Y on a corexy machine for homing both axes?

fizo666, 2021-06-11:
so.....is there a way to put an duet 2 wify doing sensorless homing?

lukeslaboratory, 2021-08-08:
Likely, Yes. @fizo666 you'll need to allow PB5 to be mapped twice using https://www.klipper3d.org/Config_Reference.html#duplicate_pin_override
and then configure the homing parameters per https://www.klipper3d.org/TMC_Drivers.html

If its an active high signal from the OR gate, make sure to pull the floating signal down with a ~

DefektSan, 2021-08-11:
> Likely, Yes. @fizo666 you'll need to allow PB5 to be mapped twice using https://www.klipper3d.org/Config_Reference.html#duplicate_pin_override
> and then configure the homing parameters per [...]

Not working.
I'm tested what heppend with pins voltage on mcu PB5 and driver SG_TST when homing.

On RRF after power on board SG_TST = 0v, move with no home SG_TST = 3.3v, move when axis go home SG_TST = 0v, and SG_TST = 3.3v when axis pushd limits and RRF homed axis successful, move away from limit, powerf off motors, move again with no home.
All this time PB5 = 1.5v.

On klipper same situation with SG_TST pin voltage, but PB5 = 0.15v because we use pull down ~

Concluding from the information above, mcu pin PB5 does not get 3.3v from driver pin SG_TST.
Mcu pin PB5 always stay 1.5v.
Klipper standard method for sensorless homing on duet board never be work.
Need another way for sensorless homing.

3v0k, 2021-09-15:
hello
i'm trying the sensorless homing for my duet2 with PB5 on endstop_pin
but same result, he's not homing however i can move the axis after "home"