Klipper3d/klipper #3299: Is warning for Linear Advance also valid for pressure advance

Twilek-de, 2020-09-08:
I habe a general question. As I understand Linear Advance is pretty similar to Pressure Advance. There is a warning to not use Linear Advance on Trinamic drivers. I just want to know wether this is also true for Pressure Advance before I inadvertently fry my drivers.

```Trinamic (TMC) Driver Notes: On any board running TMC drivers it is highly recommended to set the E driver mode to SpreadCycle and NOT StealthChop. Linear Advance uses very quick and hard moves on the E motor which can lead to lost steps when using StealthChop. Disable StealthChop for your E driver if you have a board with TMC drivers on it (EZBoard, SKR, etc). Per Trinamics' own datasheets you should not be using StealthChop on an E motor at all due to the low torque output for that mode. The EZBoard works with Linear Advance in either mode but it is still recommended to use SpreadCycle so you have proper torque for the E motor.

Machine/Board Compatibility Notes: Printers that have Trinamic (aka TMC) drivers that have no firmware control to set the driver chip into SpreadCycle and/or have poor thermal heatsinking will not be able to use Linear Advance. Examples are the CR-10S Pro/Max, CR-10S V2/V3, and all of the Creality "Silent" boards that have TMC drivers on them. These printers and/or boards will not work with Linear Advance due to the limitations of these printers/boards.```

KevinOConnor, 2020-09-08:
Klipper implements a "smooth pressure advance" system that does not result in instantaneous velocity changes: https://www.klipper3d.org/Kinematics.html#pressure-advance

It is hard to say whether or not stealthchop will work well on your particular printer.  Some people use it; some people get better results with spreadcycle.  If in doubt, go with spreadcycle.

I'm not aware of reports that using stealthchop results in damage.  Typically the symptoms are low torque and/or high heat.

-Kevin

Hywelmartin, 2020-09-08:
First... stealtchop is torquier(word) in most cases https://youtu.be/GVs2d-TOims?t=954 especially at the lower part of the RPM interval.. 

Second.... Tim Hooligans advice is best served with a ship load of salt... at least that is may take....

I have tested a lot of combinations (TMC2130) and haven't really seen any difference.. at normal speeds...

NON Uart/SPI TMC might run into the phenomenon that Klipper struggled with before...  and obviously Marlin struggles with (don't care??) like creality 1.1.5 boards

Twilek-de, 2020-09-09:
Thanks for the info!

closed 2020-09-09 by Twilek-de