Klipper3d/klipper #3301: Klipper Exception: Unable to build C code module (error=256)

DmA81, 2020-09-08:
Hi, on an updated raspbian (testing) klipper does not start anymore (fresh git pull from -master):

**klippy.log:**
```
Starting Klippy...
Args: ['/opt/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-779-g91979756'
CPU: 4 core ?
Python: '2.7.18 (default, Apr 20 2020, 20:30:41) \n[GCC 9.3.0]'
Building C code module c_helper.so
Unable to build C code module (error=256)
```

**Launching manually:**
```
$ /home/pi/klippy-env/bin/python /opt/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
during IPA pass: icf
/opt/klipper/klippy/chelper/pyhelper.c:94:1: internal compiler error: Errore di bus
   94 | }
      | ^
0xf75c9cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/serialqueue.c:1133:1: internal compiler error: Errore di bus
 1133 | }
      | ^
0xf7670cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/stepcompress.c:663:1: internal compiler error: Errore di bus
  663 | }
      | ^
0xf7609cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/itersolve.c:301:1: internal compiler error: Errore di bus
  301 | }
      | ^
0xf79a0cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/trapq.c:177:1: internal compiler error: Errore di bus
  177 | }
      | ^
0xf781bcdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/kin_cartesian.c:51:1: internal compiler error: Errore di bus
   51 | }
      | ^
0xf7312cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: cp
/opt/klipper/klippy/chelper/kin_corexy.c:40:1: internal compiler error: Errore di bus
   40 | }
      | ^
0xf759ccdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: cp
/opt/klipper/klippy/chelper/kin_corexz.c:40:1: internal compiler error: Errore di bus
   40 | }
      | ^
0xf771bcdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: cp
/opt/klipper/klippy/chelper/kin_delta.c:41:1: internal compiler error: Errore di bus
   41 | }
      | ^
0xf7973cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/kin_polar.c:59:1: internal compiler error: Errore di bus
   59 | }
      | ^
0xf79eecdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/kin_rotary_delta.c:73:1: internal compiler error: Errore di bus
   73 | }
      | ^
0xf7313cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/kin_winch.c:42:1: internal compiler error: Errore di bus
   42 | }
      | ^
0xf7328cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/kin_extruder.c:135:1: internal compiler error: Errore di bus
  135 | }
      | ^
0xf796bcdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
during IPA pass: icf
/opt/klipper/klippy/chelper/kin_shaper.c:409:1: internal compiler error: Errore di bus
  409 | }
      | ^
0xf7587cdf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
Traceback (most recent call last):
  File "/opt/klipper/klippy/klippy.py", line 330, in <module>
    main()
  File "/opt/klipper/klippy/klippy.py", line 315, in main
    printer = Printer(bglogger, start_args)
  File "/opt/klipper/klippy/klippy.py", line 54, in __init__
    self.reactor = reactor.Reactor()
  File "/opt/klipper/klippy/reactor.py", line 257, in __init__
    SelectReactor.__init__(self)
  File "/opt/klipper/klippy/reactor.py", line 99, in __init__
    self.monotonic = chelper.get_ffi()[1].get_monotonic
  File "/opt/klipper/klippy/chelper/__init__.py", line 215, in get_ffi
    , OTHER_FILES)
  File "/opt/klipper/klippy/chelper/__init__.py", line 203, in check_build_code
    raise Exception(msg)
Exception: Unable to build C code module (error=256)
```

**gcc --version:**
`gcc (Raspbian 10.2.0-5+rpi1) 10.2.0`

Anyone with the same issue ?

Thanks

KevinOConnor, 2020-09-11:
These types of errors are usually the result of a bad OS installation, and/or a failing SD card.  I recommend a complete reinstall of the sdcard.

-Kevin

DmA81, 2020-09-12:
Yep, among other problems I reinstalled raspbian from scratch and klipper now works again...

Thanks for your work !

closed 2020-09-12 by DmA81