Klipper3d/klipper #3302: Strange X and Y behavior after turning on of the printer.

vladimir-poleh, 2020-09-08:
I have CoreXY printer based on the MKS Gen L v2.0. Drivers on the X and Y are TMC2209, Z and extruder - TMC2100. TMC2209 drivers are running in the UART mode.

After `FIRMWARE_RESTART` command everything is working well but if I'm turning off my printer and turning it on I have doubled distance for the G1 commands on the X and Y axes. For example command `G1 X10 F5000` will move head to 20mm instead of 10mm. At the same time Z is working well.

I have tried to change `microsteps` parameter for the TMC2209 drivers but behavior is the same. With TMC2100 on the X and Y everything is working fine.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5190555/klippy.log)

Hywelmartin, 2020-09-08:
probably a power on sequence fault

confirme that the steppers are connected
DUMP_TMC STEPPER=<name>: This command will read the TMC driver registers and report their values.

if not run
INIT_TMC STEPPER=<name>: This command will intitialize the TMC registers. Needed to re-enable the driver if power to the chip is turned off then back on.

a stepperprotector v2 should get the correct power on sequence...

KevinOConnor, 2020-09-11:
If you remove motor power from trinamic drivers then (at a minimum) it will be necessary to run RESTART after reapplying power - it is not valid to remove motor power to the trinamic drivers while they still receive power from their digital interfaces.

-Kevin

vladimir-poleh, 2020-09-11:
Thank you for explanation. I think that this issue can be closed.

closed 2020-09-22 by KevinOConnor