Klipper3d/klipper #3305: Klipper stops/crashes middle of print

fietstasss, 2020-09-09:
Hi all,
I'm having issues with my 3d printer stopping/crashing somewhere in the middle of a print.  I've tried the same print (same gcode) for a few times and it stops -about- at the same spot every time, not exactly the same point though. 
I've also tried with OctoPrint running in safe mode and also tried with another one in safe-mode and different gcode which had lower speeds (50mm/s), all still concluded in the same behaviour as described above. 

Octoprint log seems not very helpfull, here's a snippet of the time of crashing (i think): 

> `2020-09-08 21:01:50,448 - octoprint.plugin - DEBUG - Calling on_event on klipper
> 2020-09-08 21:02:00,370 - octoprint.timelapse - DEBUG - Capturing image to /home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-161.jpg
> 2020-09-08 21:02:00,372 - octoprint.timelapse - DEBUG - Going to capture /home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-161.jpg from http://192.168.30.250/webcam/?action=snapshot
> 2020-09-08 21:02:00,372 - octoprint.events.fire - DEBUG - Firing event: CaptureStart (Payload: {'file': u'/home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-161.jpg'})
> 2020-09-08 21:02:00,375 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xac54d050>>
> 2020-09-08 21:02:00,379 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
> 2020-09-08 21:02:00,382 - octoprint.plugin - DEBUG - Calling on_event on file_check
> 2020-09-08 21:02:00,383 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
> 2020-09-08 21:02:00,384 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
> 2020-09-08 21:02:00,385 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
> 2020-09-08 21:02:00,386 - octoprint.plugin - DEBUG - Calling on_event on DryRun
> 2020-09-08 21:02:00,386 - octoprint.plugin - DEBUG - Calling on_event on IFTTT
> 2020-09-08 21:02:00,388 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
> 2020-09-08 21:02:00,389 - octoprint.plugin - DEBUG - Calling on_event on dashboard
> 2020-09-08 21:02:00,390 - octoprint.plugin - DEBUG - Calling on_event on enclosure
> 2020-09-08 21:02:00,390 - octoprint.plugin - DEBUG - Calling on_event on excluderegion
> 2020-09-08 21:02:00,391 - octoprint.plugin - DEBUG - Calling on_event on filamentmanager
> 2020-09-08 21:02:00,391 - octoprint.plugin - DEBUG - Calling on_event on klipper
> 2020-09-08 21:02:00,430 - octoprint.timelapse - DEBUG - Image /home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-161.jpg captured from http://192.168.30.250/webcam/?action=snapshot
> 2020-09-08 21:02:00,431 - octoprint.events.fire - DEBUG - Firing event: CaptureDone (Payload: {'file': u'/home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-161.jpg'})
> 2020-09-08 21:02:00,432 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xac54d050>>
> 2020-09-08 21:02:00,434 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
> 2020-09-08 21:02:00,434 - octoprint.plugin - DEBUG - Calling on_event on file_check
> 2020-09-08 21:02:00,435 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
> 2020-09-08 21:02:00,436 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
> 2020-09-08 21:02:00,437 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
> 2020-09-08 21:02:00,437 - octoprint.plugin - DEBUG - Calling on_event on DryRun
> 2020-09-08 21:02:00,438 - octoprint.plugin - DEBUG - Calling on_event on IFTTT
> 2020-09-08 21:02:00,440 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
> 2020-09-08 21:02:00,441 - octoprint.plugin - DEBUG - Calling on_event on dashboard
> 2020-09-08 21:02:00,441 - octoprint.plugin - DEBUG - Calling on_event on enclosure
> 2020-09-08 21:02:00,443 - octoprint.plugin - DEBUG - Calling on_event on excluderegion
> 2020-09-08 21:02:00,443 - octoprint.plugin - DEBUG - Calling on_event on filamentmanager
> 2020-09-08 21:02:00,444 - octoprint.plugin - DEBUG - Calling on_event on klipper
> 2020-09-08 21:02:10,371 - octoprint.timelapse - DEBUG - Capturing image to /home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-162.jpg
> 2020-09-08 21:02:10,374 - octoprint.timelapse - DEBUG - Going to capture /home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-162.jpg from http://192.168.30.250/webcam/?action=snapshot
> 2020-09-08 21:02:10,375 - octoprint.events.fire - DEBUG - Firing event: CaptureStart (Payload: {'file': u'/home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-162.jpg'})
> 2020-09-08 21:02:10,378 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xac54d050>>
> 2020-09-08 21:02:10,392 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
> 2020-09-08 21:02:10,393 - octoprint.plugin - DEBUG - Calling on_event on file_check
> 2020-09-08 21:02:10,395 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
> 2020-09-08 21:02:10,400 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
> 2020-09-08 21:02:10,404 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
> 2020-09-08 21:02:10,412 - octoprint.plugin - DEBUG - Calling on_event on DryRun
> 2020-09-08 21:02:10,413 - octoprint.plugin - DEBUG - Calling on_event on IFTTT
> 2020-09-08 21:02:10,417 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
> 2020-09-08 21:02:10,419 - octoprint.plugin - DEBUG - Calling on_event on dashboard
> 2020-09-08 21:02:10,421 - octoprint.plugin - DEBUG - Calling on_event on enclosure
> 2020-09-08 21:02:10,424 - octoprint.plugin - DEBUG - Calling on_event on excluderegion
> 2020-09-08 21:02:10,427 - octoprint.plugin - DEBUG - Calling on_event on filamentmanager
> 2020-09-08 21:02:10,431 - octoprint.plugin - DEBUG - Calling on_event on klipper
> 2020-09-08 21:02:10,558 - octoprint.timelapse - DEBUG - Image /home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-162.jpg captured from http://192.168.30.250/webcam/?action=snapshot
> 2020-09-08 21:02:10,561 - octoprint.events.fire - DEBUG - Firing event: CaptureDone (Payload: {'file': u'/home/pi/.octoprint/timelapse/tmp/068_CarClip_20200908203509-162.jpg'})
> 2020-09-08 21:02:10,561 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xac54d050>>
> 2020-09-08 21:02:10,563 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
> 2020-09-08 21:02:10,564 - octoprint.plugin - DEBUG - Calling on_event on file_check
> 2020-09-08 21:02:10,567 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
> 2020-09-08 21:02:10,569 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
> 2020-09-08 21:02:10,571 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
> 2020-09-08 21:02:10,572 - octoprint.plugin - DEBUG - Calling on_event on DryRun
> 2020-09-08 21:02:10,573 - octoprint.plugin - DEBUG - Calling on_event on IFTTT
> 2020-09-08 21:02:10,575 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
> 2020-09-08 21:02:10,576 - octoprint.plugin - DEBUG - Calling on_event on dashboard
> 2020-09-08 21:02:10,577 - octoprint.plugin - DEBUG - Calling on_event on enclosure
[... 118 lines cut ...]
> 2020-09-08 21:02:12,454 - octoprint.plugin - DEBUG - Calling on_event on dashboard
> 2020-09-08 21:02:12,457 - octoprint.plugin - DEBUG - Calling on_event on enclosure
> 2020-09-08 21:02:12,498 - octoprint.plugin - DEBUG - Calling on_event on excluderegion
> 2020-09-08 21:02:12,499 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelled
> 2020-09-08 21:02:12,506 - octoprint.plugin - DEBUG - Calling on_event on filamentmanager
> 2020-09-08 21:02:12,506 - octoprint.events - WARNING - Command failed with return code 2, enable debug logging on target 'octoprint.events' for details
> 2020-09-08 21:02:12,545 - octoprint.plugin - DEBUG - Calling on_event on klipper
> 2020-09-08 21:02:12,566 - octoprint.events.fire - DEBUG - Firing event: PrintCancelling (Payload: {'origin': u'local', u'firmwareError': u"Lost communication with MCU 'mcu'", 'name': u'068_CarClip.gcode', u'owner': 'fietstasss', 'path': u'068_CarClip.g [... cut ...]
> 2020-09-08 21:02:12,716 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Lost communication with MCU 'mcu'
> | Last lines in terminal:
> | Recv: // config, and restart the host software.
> | Recv: // Printer is shutdown`

Attached you can also find my klippy log.  
I've tried to 'extract information' with the python command, but didn't succeed in this. 
(as described here https://www.klipper3d.org/Debugging.html#:~:text=The%20Klippy%20log%20file%20(%2F,these%20statistics%20after%20a%20print.&text=One%20can%20then%20view%20the%20resulting%20loadgraph.)

BlackStump, 2020-09-09:
That is way tooooo busy for my eyes. please do not copy and paste but attach as a log
where is the klippy.log

fietstasss, 2020-09-09:
Fixed with edit, excuse me.

KevinOConnor, 2020-09-11:
See the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

closed 2020-10-11 by github-actions[bot]