Klipper3d/klipper #3310: Z Probe / first layer height question - how to automate?

Tahutipai, 2020-09-10:
For the purpose of this question, assume a perfectly flat and level print bed.
Printer bed homes away from the nozzle, for safety.  Endstop_pos=300

I have a custom built Z Probe that works well, range <0.0125m for N=10.  Prob uses the nozzle tip itself, there is a slight deflection, so probe is "above" the nozzle (z_offset=-0.4)

Question: how to automate probing & gcode offset adjustment?  I feel I am missing something.

Currently:
-I manually run a ``PROBE``, lets say this reports actual Probe Height is at Z2.5mm
-I then manually add the probe z_offset , and manually run SET_GCODE_OFFSET Z=2.9mm
(I guess, alternatively, I could also manually adjust z endstop_pos, so that nozzle height becomes at Z=0.0

And this works perfectly.  But I think it's wrong - surely there must be some way to automatically probe the ZHeight / update the offset?  I would like to do this automatically every print.

Question: should I be able to do this automatically with Klipper, or must it be done manually?  Is my process wrong?

Tahutipai, 2020-09-10:
Thanks bot, log not required for this ticket. Issue not printer dependent or runtime error.

KevinOConnor, 2020-09-11:
It's possible to alter the g-code coordinate system using the SET_GCODE_OFFSET and/or G92 commands.  You could thus create a gcode_macro that runs the probe and then updates the g-code coordinate system accordingly.

-Kevin

closed 2020-09-15 by Tahutipai