Klipper3d/klipper #3312: Octoprint: Upgrade to Python 3 recommended

top-gun, 2020-09-11:
My Octopi 0.17 is based on Python 2, everything is fine. The maintainers of Octoprint just released a recommendation to upgrade installations of Octopi 0.17 to Python 3, and that 0.18 will be based on Python 3 out of the box.

There is a prepared script to install Python 3 on the octopi.

https://octoprint.org/blog/2020/09/10/upgrade-to-py3/?utm_source=octoprint&utm_medium=announcements&utm_content=1.4.2

I don't know much about Python, I just installed an Octopi and installed Klipper following the documentation. The installation pulls from "master" and was updated last week.

If I use the script, will this break my Klipper-installation? I got advice in the blog-comments that Klipper should run "in its own virtual environment". My knowledge of Python does not go that far.

Thanks for any advice.

swenox, 2020-09-11:
I´m not an expert in the field but i have performed the update on one of my machines. 
For me everything still working fine except octoklipper plugin that still dosent support Python3.

KevinOConnor, 2020-09-11:
OctoPrint is a separate project - its choicse do not impact Klipper.  The OctoKlipper plugin is a separate project that is not developed here.

-Kevin

vladimir-poleh, 2020-09-11:
Klipper and OctoPrint are different apps with own Python environments so it is safe to upgrade OctoPrint to Python 3. Please note that OctoKlipper plugin was not updated to the Python 3 but there is a fork for this plugin.

DanielJoyce, 2020-09-11:
Python2 is no longer getting fixes, so eventually it will affect Klipper....

top-gun, 2020-09-11:
> Klipper and OctoPrint are different apps with own Python environments so it is safe to upgrade OctoPrint to Python 3. Please note that OctoKlipper plugin was not updated to the Python 3 but there is a fork for this plugin.

Thank you, that's what I needed to know. I understand that installing Klipper on Octopi 0.18 might take some additional steps, but that's a bridge I will need to cross in the future, like most others.

@ Kevin: I understand your point. Yet, Octopi is a supported and common foundation for Klipper, and it was not clear to me if I can follow the recommendation from the Octopi maintainers.

closed 2020-09-14 by top-gun