Klipper3d/klipper #3313: TMC2208 noisy with master branch

FlashyM, 2020-09-11:
I have Klipper running on several printers but there are problems on a cartesian printer with MKS Gen L 1.0 board and TMC2208 stepsticks. UART is connected. Everything is running very smooth and silent with branch "abdfd92". After switching to master branch all steppers are more or less noisy with unchanged config file. It's not a constant noise, sometimes I can't hear anything and sometimes it sounds like damaged mechanics. The printer does not lose steps though.
I tried all possible TMC2208 config settings and could reduce noise on z-axis by setting microsteps to 64 but this didn't help much on x- and y-axis.
Any idea - what's different between master branch and "abdfd92"?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5211556/klippy.log)
?

paul-arnold, 2020-09-12:
As for changes between  abdfd92 and the master branch, I would do a simple diff.
Can you post a dump_tmc output of the steppers in question?

FlashyM, 2020-09-14:
Here are the TMC dumps of x-, y- and z-axis after homing and moving up (G28 X Y Z, G1 Z30), with both master branch (noisy) and branch abdfd92 (smooth, quiet and flawless). 
[dump_tmc.zip](https://github.com/KevinOConnor/klipper/files/5218707/dump_tmc.zip)

FlashyM, 2020-09-15:
Ok, last night I did plenty of tests through the commit history and I guess I nailed it down:

1.) Crackling noises on x- and y-steppers (and perhaps also on extruder stepper) started along with commit 271e412 from March 5th. There were no issues with the previous commit bc6c3ba.

2.) The additional grinding noise on the z-stepper started along with commit ea5e767 from June 9th. The crackling noises on x- and y-steppers got more and louder with this commit.

The crackling noises on x- and y-steppers are appearing mainly in curves, e. g. in the curved skirt when printing my first layer test. Skirt is printed with 25 mm/sec.

Btw., I got plenty of "TIMER TOO CLOSE" errors, mainly when restarting the klipper service.

KevinOConnor, 2020-09-15:
Interesting.  Both of those commits can make minor changes to the timing of steps - however, neither timing change would be "better" or "worse" - just a little different.

My guess is that your setup was "on the edge" before and minor timing differences are now making it more noticeable.

Some recommendations:
1. Make sure your stealthchop_threshold is either zero (to use spreadcycle) or way above the maximum possible velocity for the axis (to use stealthchop).  It is now okay to set stealthchop_threshold to 99999 to always use stealthchop.
2. Return to the default square_corner_velocity of 5mm/s, unless you have a specific requirement to use a higher value.  The tmc drivers really don't like instantaneous velocity changes in stealthchop mode.
3. Set your X and Y step distance to .0125 .  Your current customized settings don't look correct (they would indicate a uniformly stretched belt which is highly unlikely).  The "unusual values" that you've set will result in micro speed jumps during common constant velocity moves (because there wont be a whole number of micro-controller clock ticks per step with the "unusual" step-distance setting).
4. Try tuning your stepper current settings.  A higher or lower current will alter the heat and sound of the drivers.

-Kevin

FlashyM, 2020-09-15:
Thanks for your response. Tested again with your recommended (default) settings and different driver voltage. Unfortunately this didn't help, the noise is still there, unchanged. Logfile is attached.
I have the impression the steppers are doing some big steps between the normal ones.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5226642/klippy.log.zip)

KevinOConnor, 2020-09-15:
That's certainly odd.  Is the host machine using a 32bit OS or a 64bit OS?

Do you have a short sequence of g-code commands that exhibit the problem?  If you can come up with a sequence it will be possible to compare the micro-controller commands that are produced between the different revisions.

-Kevin

FlashyM, 2020-09-15:
Host is a 64-bit-Windows 10. The problem appeared already before on a Raspberry Pi, I've prepared this virtual machine for faster testing. You'll find a simple first-layer-test attached.
[First-Layer-Test-PLA.zip](https://github.com/KevinOConnor/klipper/files/5226973/First-Layer-Test-PLA.zip)

, I hear the z-stepper already when homing and the x-/y-steppers when printing the skirt.

FlashyM, 2020-09-15:
For clarification:
Klipper is running on newest 32-bit Raspberry OS (Debian) in VirtualBox.

KevinOConnor, 2020-09-15:
There were problem reports in the past with old X86 80bit floating point numbers vs 64bit floating point numbers.  The newer 64bit X86 OS images use the standard 64bit floating point, but the 32bit images will typically use 80bit math.  See #2198 .  It's probably worth while to check that first before digging further into the kinematics.  (Either verify on an rpi, spin up a 64bit OS, or make sure the compiler flags avoid 80bit math.)

As for tracking down kinematic changes, it would help if you could reduce the sequence of commands to a smaller set.  Once you have a sequence, you can follow the directions at: https://www.klipper3d.org/Debugging.html#translating-gcode-files-to-micro-controller-commands to see what each version of the code is doing.

-Kevin

KevinOConnor, 2020-09-16:
I don't know if it is related to this issue or not, but I did update the compile process to avoid 80bit floating point math on x86 hosts (commit a92d9957).

-Kevin

FlashyM, 2020-09-16:
Great, the compiler flags did the trick, printing smooth and quiet now!
Now I'm finally able to test and hopefully use the new input shaper.
Thanks a lot!

closed 2020-09-16 by FlashyM