Klipper3d/klipper #3316: MCU 'mcu' shutdown: Timer too close

bearded1der, 2020-09-12:
Experienced an issue I've not seen in the past on a long print last night. Awoke this morning to find the printer in shutdown state, midway through, klipper was reporting repeatedly "MCU 'mcu' shutdown: Timer too close"

I found some past issues on this problem related to overtaxing the MCU, mainly issue #288 - reported closed. I followed the steps to generate a graph of the operations, as in that thread, as well as looked in the log manually, and the parsed shutdown log, but I'm unsure how to further interpret these to find the issue. 

I did recently set up and tune input shapers, and resulting raised print speeds and acceleration, but have had several successful prints since then with the same slicer settings, so I'm doubtful this is the issue. Any further guidance would be greatly appreciated.

sergiocipo, 2020-09-15:
your heatedbed failed to mantain the terget temperature. The target temperature was 55 and the readings was 36. Check your heatedbed

pandaym3D, 2020-09-15:
Did you figure this out @bearded1der ?

KevinOConnor, 2020-09-15:
It looks like something in the input g-code caused the host machine to become overloaded - likely due to additional load caused by input_shaper.

What type of host machine are you using?

Can you provide the g-code file that was printing at the time of the failure?

@dmbutyugin - FYI.

-Kevin

bearded1der, 2020-09-16:
@KevinOConnor thanks for the reply. Attaching GCODE [HERE](https://github.com/KevinOConnor/klipper/files/5229057/core-c-primo_0.28mm_PLA_ENDER3_19h6m.zip).
 I have not yet attempted to recreate the issue with the same file, as the last wasted approximately a 1/4 kilo of material and I didn't want to repeat it if it could be helped. It is a hefty file, with a rather solid Gyroid infill and fairly high step rate. The infill was the only real change on the slicing side from a previous file that completed successfully after a significantly longer elapsed print time. Performance up to the stop point was flawless, it's a little hard to fathom that it would eventually bog down after repeating the same pattern for 7 hours previously without issue.

Klipper is running under Octoprint on an original Pi 3B+. I've been running Klipper itself since sometime last year, but obviously just enabled input-shapers a few prints before. There are plenty of other add-ons on my Octoprint instance, including a running Octolapse timelapse during the failed print, but none of them new or changed to speak of. Octolapse was also running on the previous successful print. Actual printer is a modified Ender3 on the stock ATMEGA 1284P board.

If there's any other info I can dig up or testing I can do, I'm all ears!

@pandaym3D I've not been having consistent trouble with mine yet, just had one failure thus far, but wanted to head it off before I got too deep in another big print.

KevinOConnor, 2020-09-17:
I've updated the code to improve the handling of "python garbage collection" (primarily commit cb0a8f2). Anyone that is getting "Timer too close" errors should update to the latest code (see the FAQ for upgrade instructions). It is not clear if this will resolve the issue - if you continue to encounter these errors, please report them (along with the associated klipper log).

-Kevin

dmbutyugin, 2020-09-17:
@KevinOConnor, to wrap up the conversation from another thread:
> in contrast, issue #3316 shows Klipper trying to generate steps when it eventually runs out of time.

I actually ran the timing measurements of the lookahead planner and _process_moves method yesterday using the gcode posted by bearded1der (and with his config) with pretty much the same results. The longest runtime of _process_moves is 20-30 msec on RPi 3b+, and lookahead planner is even faster. Thus I also do not think that in this case the iterative solver gets stuck or anything like it. So it does make sense to try the latest improvements of garbage collection, hopefully it should resolve 'Timer too close' issue.

KevinOConnor, 2020-09-18:
@dmbutyugin - Thanks for investigating.  I agree that input_shaper is unlikely to be the culprit - other than by, perhaps, increasing system load a little.  Sorry for my earlier comments on input_shaper.  Hopefully the recent "garbage collection" changes will resolve the underlying issue.

-Kevin

dmbutyugin, 2020-09-18:
@KevinOConnor No, I think it was a good idea to check if input_shaper could cause an issue in this case - after all, it did cause some issues in iterative solver in the past. Now since we know it is unlikely to be a culprit, we can be more confident that, perhaps, garbage-collection improvements will help in this case too.

Separately, I think input_shaper can indirectly contribute to a certain fraction of these issues. Namely, after enabling input_shaper, users can typically print faster, with higher velocity and higher accelerations. This can directly increase the required step rate for the extruder due to pressure advance (higher the acceleration - higher the PA extruder velocity). And higher average print speeds (due to higher acceleration and velocity) will require higher rate of processing of moves. So if some fractions of systems were near the boundary, after which the issues start, ironically input_shaper can trip them over that boundary. It does not mean it is a root cause in this case, of course, even if it looks like it.

bearded1der, 2020-09-19:
Thanks for the input and work on this guys. I will pull the latest code and give it a shot.

Out of curiosity, do you think it would be worthwhile to run a "processing torture test" type print for further testing of the issue under real world condition communicating with the mcu, or is that redundant with the debug mode testing that's been done already?

KevinOConnor, 2020-09-22:
It's extremely difficult to simulate real-world micro-controller behaviour.  The current test cases try to "exercise the code", but they are unlikely to catch all errors.  For this particular incident, I would continue to print as normal and report if the error returns.

-Kevin

closed 2020-10-21 by github-actions[bot]

DroneMang, 2021-01-13:
Having the same issue today, just started yesterday after changing the bed mesh settings.
[klippy8.zip](https://github.com/KevinOConnor/klipper/files/5806080/klippy8.zip)