Klipper3d/klipper #3318: TemplateSyntaxError: expected token 'end of print statement', got ':'

dertbv, 2020-09-13:
Klipper will not start giving me the error message.  i have attached the log and the menu.cfg. i can figure out where the issue is..

Error loading template 'menu __tune __speed:name': TemplateSyntaxError: expected token 'end of print statement', got ':'

[klippy.log](https://github.com/KevinOConnor/klipper/files/5213434/klippy.log)

[menu.txt](https://github.com/KevinOConnor/klipper/files/5213435/menu.txt)renamed to allow to upload

Stetinac, 2020-09-15:
Hello,

 I have the same issue as you. When I try to disable "Tune menu" Klipper print:  _error with !! Error loading template 'menu __control __fanonoff:name': TemplateSyntaxError: expected token 'end of print statement', got ':_

Problem is when "name" contains variables - example: name: Fan speed: **{1:3d}**
Same issue with variables have "gcode"

KevinOConnor, 2020-09-15:
Be sure to follow the directions in the FAQ when upgrading the software: https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

In particular, be sure to read and follow the instructions in the Config Changes document: https://www.klipper3d.org/Config_Changes.html .  The menu system has changed and all custom menus need to be updated.

-Kevin

closed 2020-09-18 by dertbv