Klipper3d/klipper #3319: Z Offset

amelios, 2020-09-13:
It seems like there's been some sort of change between 0.8.0 and master that would change the behavior of z offset with respect to mesh probing. With 0.8.0 (and some, but not super recent releases), as part of my start sequence, I'd do a g28 and a bed_mesh_calibrate, and things would be good. When I do the same with what's on master at the moment, the nozzle makes contact with the bed when I command it to a z of about 2 (which obviously leads to unfortunate circumstances when commanded lower). It seems like if I do another G28 after that sequence it seems fine (commanding to a Z of 2 puts me roughly 2mm above the print surface).

Yes, I've done and saved and restarted after doing a PROBE_CALIBRATE. The z_offset computed there is equivalent between the releases.

As a workaround for now, I've simply added another G28 after my sequence, but obviously it'd be nice to not have to do this.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5213628/klippy.zip)

amelios, 2020-09-16:
I believe this may be related to #3324 and #3334

Arksine, 2020-09-17:
I'll take a look at this.  I'm not able to reproduce locally, so I will need some more info.   If you could reproduce with the following commands and upload the resulting klippy.log it would be helpful:
1) G29
2) GET_POSITION
3) G1 X30 Y30 Z3
4) GET_POSITION

The GET_POSITION commands will log additional information that is useful in troubleshooting this kind of issue.

amelios, 2020-09-18:
See logs attached! When I powered up, I had the extra G28 in my G29 gcode sequence. I did nothing except change the config to remove that, restarted and did the sequence above. I think readded the G28 and did this sequence.

I have to say, I was somewhat surprised by what I saw after the first sequence. It seemed like it was reasonably far above the bed (seemed like it was definitely more than a millimeter, which is what I would've expected based on my past experience)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5247467/klippy.zip)
.

Arksine, 2020-09-19:
Looking at that log it seems the bed_mesh module is functioning as expected.  Its adding a bit of  Z adjustment when you move to X30 Y30 Z3, rather than subtracting it which is what I would expect if it was the cause of your issue.   Perhaps it needs to move to Z2 in order to reproduce?

If you are able to reproduce this behavior again run GET_POSITION and upload the resulting log.

amelios, 2020-09-19:
I'll see if I can get some new logs soon. In the meantime, I came across this ticket (which I haven't read in its entirety yet), but am wondering if it's somehow related: #2509

Twilek-de, 2020-10-07:
I have seen that there is a new developement branch titled "BLTouch". Will this problem here be adressed as well? I had to stop using Klipper because of [https://github.com/KevinOConnor/klipper/issues/3334](url) and would be really happy to come back if I could have a stable Z Offset (as Marlin is providing at the moment).

amelios, 2020-10-30:
Any chance this was cleared up by c2c79ff98d6f5ff0305d69afbbb72a7113ba6251 ?

I realized that between versions (where I'd seen and not seen the problem), I'd also said stow_on_each_sample to false

KevinOConnor, 2020-11-09:
That bug impacted the "stow_on_each_sample=False" config - which the original log shows some uses of.  So, it is possible.

-Kevin

closed 2020-12-08 by github-actions[bot]