Klipper3d/klipper #3321: KLIPPY_ARGS in /etc/init.d/klipper are not picked up by service

apphead, 2020-09-13:
Apparently, the KLIPPY_ARGS have no impact (using current version as of Sept 13 2020 / 063f9a2c)

Compare path to printer.cfg and additional param  -a /tmp/klippy_uds in init.d with the one in klippy.log.

```
Configuration for /etc/init.d/klipper
KLIPPY_USER=pi

KLIPPY_EXEC=/home/pi/klippy-env/bin/python

KLIPPY_ARGS="/home/pi/klipper/klippy/klippy.py /home/pi/klipper_config/printer.cfg -l /tmp/klippy.log -a /tmp/klippy_uds"
```

this is what I get in klippy.log:

```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-786-g063f9a2c'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Start printer at Sun Sep 13 14:27:02 2020 (1600000022.9 771.3)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 104, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 146, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 132, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 221, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 110, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg
```
I used installer script https://github.com/th33xitus/kiauh/tree/master with Mainsail and Moonraker up and running.
Is there another way to start the klipper service that uses different ARGS? What am I missing?

theopensourcerer, 2020-09-13:
I have this working just fine. As I recall the only file I changed was /etc/default/klipper, not /etc/init.d/klipper as described in the mainsail installation instructions:

https://github.com/meteyou/mainsail/blob/master/docs/installation.md

theopensourcerer, 2020-09-13:
Yep - this is the script that becomes init.d on installation:

https://github.com/KevinOConnor/klipper/blob/master/scripts/klipper-start.sh#L23

apphead, 2020-09-13:
@theopensourcerer init.d was a typo, I changed /etc/default/klipper of course which is the one I listed above. But no matter what I put in KLIPPY_ARGS it doesn't change how the service is started.

apphead, 2020-09-13:
Funnily, when I start the service like that
`~/klipper/scripts $ sudo ./klipper-start.sh start`

klippy.log shows the correct args:
`Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/tmp/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.8.0-786-g063f9a2c'`

apphead, 2020-09-13:
It's getting real frustrating. I know something is off here, but my linux is just not good enough to understand how to fix it.
Facts so far:
KLIPPY_ARGS in /etc/default/klipper are ignored by the daemon
I removed /etc/default/klipper and started & stopped klipper service and the log file looks exactly the same. WTF!?!?

theopensourcerer, 2020-09-14:
I think if I were you I'd start again doing a fresh install of Klipper on a clean Raspbian and then follow the instructions on the mainsail page to install Klipper, moonraker and mainsail.

Maybe these custom scripts you linked to originally (KIAUH) have made some changes to the way the init.d script works?

It's really not very hard to install klipper and then mainsail just following the basic instructions. And as I did this just a few days ago I can say it worked fine for me.

https://github.com/meteyou/mainsail/blob/master/docs/installation.md

apphead, 2020-09-14:
I'm thinking along the same lines. But still, I'd love to understand what screwed up the service. Strangely, I seem to be the only one having problems (or people just use the default path /home/pi/printer.cfg ). Oh, well.

I did the manual installation after using KIAUH, but that did not help, so yup, maybe KIAUH is the culprit. Pitty, I do like it overall.

Will close the issue if nobody has any ideas where to look further and investigate. Thank you @theopensourcerer for you support, much appreciated!

th33xitus, 2020-09-14:
> Maybe these custom scripts you linked to originally (KIAUH) have made some changes to the way the init.d script works?
Nope, it doesn't make such changes to the init.d script. If you install Klipper through KIAUH the install-octopi.sh script of Klipper will be used to install the service. 

If you then use the Moonraker installer of KIAUH, `/etc/default/klipper` gets edited, yes. But only a single line (the one with `KLIPPY_ARGS`) and only in a very specific way.
The function i wrote for that looks as follows:
```
patch_klipper_sysfile(){
  status_msg "Checking /etc/default/klipper for necessary entries ..."
  #patching new printer.cfg location to /etc/default/klipper
  if ! grep -q "/klipper_config/printer.cfg" $KLIPPER_SERVICE2; then
    status_msg "Patching new printer.cfg location to /etc/default/klipper ..."
    sudo sed -i "/KLIPPY_ARGS=/ s|$PRINTER_CFG|/home/${USER}/klipper_config/printer.cfg|" /etc/default/klipper
    ok_msg "New location is: '/home/${USER}/klipper_config/printer.cfg'"
  fi
  #patching new UDS argument to /etc/default/klipper
  if ! grep -q -- "-a /tmp/klippy_uds" $KLIPPER_SERVICE2; then
    status_msg "Patching unix domain socket to /etc/default/klipper ..."
    #append the new argument to /tmp/klippy.log argument
    sudo sed -i "/KLIPPY_ARGS/s/\.log/\.log -a \/tmp\/klippy_uds/" $KLIPPER_SERVICE2
    ok_msg "Patching done!"
  fi
  ok_msg "Check complete!"
  echo
}
```
There are two if statements, each one checks for a specific change. The first one looks if the printer.cfg location was already updated to `/klipper_config/printer.cfg`, if not... and this is usually the case for a fresh klipper install, the default path gets replaced by the one Moonraker expects when installed through KIAUH.
The second if statement checks for the needed UDS argument... if that is not present already, the argument `-a /tmp/klippy_uds` gets appended behind the `/tmp/klippy.log` entry. No other changes are made to the init.d service configuration file and that file gets only touched if it hasn't been touched before.

I don't want to say the script is 100% perfect, because it's not. And if there are issues i am always eager to try and fix them (always feel free to report them to KIAUH's issue tracker). But at the moment i can't see an issue caused by KIAUH. Looking at the information you provided @apphead , it looks like your `/etc/default/klipper` got successfully patched. A restart if the klipper service which is done during the moonraker installation as well should apply those new starting arguments. 

A fresh Klipper install might be recommended, yes. Maybe a new clean Raspberry Pi OS Buster image. This would solve the issue probably in the most quick way. Although it won't help tracking down what causes this error.

apphead, 2020-09-14:
I started from scratch and did all the setup manually, up and running now. If I hadn't spent more than a day to figure out the issue above, I would have tested KIAUH again, but I want to move on and start printing ;-)

closed 2020-09-14 by apphead