Klipper3d/klipper #3322: mcu shutdown timer too close

sergiocipo, 2020-09-13:
Hello, I've a working installation of klipper on an sd and raspberry pi connected to SKR 1.1 PRO via USB. this configuration is perfectly working. Now I would like to test Mainsail an decided to test this on a new sd. Installed klipper and updated mcu but i get timer too close error and mcu shtutdown. I'm running the same printer.cfg on both. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5214835/klippy.log)

KevinOConnor, 2020-09-15:
The error occurred because the host machine was unusually slow during its configuration process.  What type of host machine are you using?  If you run FIRMWARE_RESTART does the error clear?

-Kevin

sergiocipo, 2020-09-15:
I'm running last raspios on Raspberry pi 3 B+. I issued an echo FIRMWARE_RESTART >/tmp/printer and same error. I installed klipper 0.7 and it looks like it works

KevinOConnor, 2020-09-16:
The log indicates only 1 cpu core was found, so if this was run on an rpi3, something is dramatically incorrect at the OS level.  I recommend you recheck the setup and operating system image.

-Kevin

sergiocipo, 2020-09-16:
Reinstalled raspy os and now it has detected correctly the cpu. klipper up and running. thank you Kevin

closed 2020-09-16 by sergiocipo