Klipper3d/klipper #3324: Mesh Bed Leveling not correcting on first layer

RaoulFabisch, 2020-09-14:
Hi,

right now I have the problem that after performing Mesh Bed Leveling the printer does not perform the correction on the first layer. I am shure I just forgot a simple thing in the config but just can't find it :)

Pls find attached the log file
[
[klippy.log](https://github.com/KevinOConnor/klipper/files/5218779/klippy.log)

](url)

KevinOConnor, 2020-09-22:
Unfortunately, there is very little anyone can do with such a vague description of an issue.  As far as we know, the code works as intended.

-Kevin

RaoulFabisch, 2020-09-23:
I will try to describe it more precise:

1. I have configured Mesh Bed leveling on my Ender with the SKR mini and BL-Touch.
2. Z-Offset is configured
3. Before every print I probe and use the following Start g-code : 
gcode =
    G28; Home all axes    
    BED_MESH_CALIBRATE
    BED_MESH_PROFILE SAVE=default
    BED_MESH_PROFILE LOAD=default    
    G92 E0; Reset Extruder
    G1 Z5.0 F3000; Move Z Axis up little to prevent scratching of Heat Bed
    G1 X3.4 Y20 Z0.3 F5000.0; Move to start position
    G1 X3.4 Y200.0 Z0.3 F1500.0 E15; Draw the first line
    G1 X3.8 Y200.0 Z0.3 F5000.0; Move to side a little
    G1 X3.8 Y20 Z0.3 F1500.0 E30; Draw the second line
    G92 E0; Reset Extruder
    G1 Z5.0 F3000; Move Z Axis up little to prevent scratching of Heat Bed

4. From Marlin I know that when performing Mesh Bed Leveling it corrects the Z-Height to correct uneveness. I now have the problem that the Z-Axis is not correcting the Z-Height and I have a good result in the middle of the print bad, but at the front it is a bit to squished and on while it is to far away at the back. The Z-Axis is not moving during the first layer which leads me to the point that I have missed something in the configuration, but can't really find out what to add there. 

I hope it is now precise enough, otherwise I can provide more information :)

Thanks in advance

ProfEngr, 2020-09-23:
TL;DR : make sure your printer.cfg has the mesh data stored at the bottom. Also, if you're going to recreate the mesh every time in start code then why use the 2 lines to store/load? Klipper will already use the result from BED_MESH_CALIBRATE if I understand correctly.

************************************************
I think you need to issue a SAVE_CONFIG (also forces a FIRMWARE_RESTART) to have the new mesh stored in printer.cfg and be available to the LOAD command.
Granted, I use an Octoklipper macro to create and store the mesh using the following steps:

M190 S50
G28
G29
M140 S0
G90
G1 X15 Y15 Z15
BED_MESH_PROFILE SAVE=MESH50C
save_config

Those steps illicit a firmware restart on their own. I have a G29 macro in printer.cfg that calls BED_MESH_CALIBRATE

I recall the mesh after G28 in my start gcode/macro using:

    BED_MESH_PROFILE LOAD=MESH50C ; PLA mesh stored end of file

Now, my printer doesn't normally need the mesh recreated unless I've bumped the knobs, change bed temps, or the 1st layer gets wonky. This can be weeks.

RaoulFabisch, 2020-09-27:
Thank you for this great answer.

I have now created a mesh profile and saved it in a correct way so that it now corrects the first layer as I expected it. Will now start to play with the fade out function and see how it works.

closed 2020-09-27 by RaoulFabisch