Klipper3d/klipper #3329: BTT EXP-MOT with TMC5160 SPI mode

Gilabite, 2020-09-14:
Hello, Im buidling a new coreXY machine with an SKR Pro 1.2 and the EXP-MOT expansion module, dont have a klipper issue need guidance instead. I want to use TMC5160 and the config only shows UART not the SPI for the exp. I think Ive got everything right except I cant find what the CS pin would be to use them in SPI. I was going to use this for my 2 other Z axis motors for 3 point bed leveling and possibly an extruder in the future. I connected the exps SPI header to the SKR Pros SPI expansion header which is tied to the moto SPI according to the schematic. It doesnt look like the EXP1 and EXP2 headers pass the CS pins that I can see. The SPI expansion header does pass the X CS pin it looks like. Thank you for anyones help.

Gilabite, 2020-09-15:
Ok so looks like I would jumper the CSN pin on the 5160 to an extra pin on the SKR? Just not sure which to use if that is right?

Gilabite, 2020-09-15:
Figured it out, I used some of the pins in expansion header 2 for the CS pin

closed 2020-09-15 by Gilabite

evandene, 2020-12-10:
> Figured it out, I used some of the pins in expansion header 2 for the CS pin

Hi,
Hope all go's well.
I'm in the process installing klipper and using TMC5160 drivers on my SKR PRO v1.2 board.
Could you please help me (read dummy) out on this matter?
you are talking about jumpers?
Also maybe you could share your printer.cfg file?
Thanks in advance
Ed van den Enden