Klipper3d/klipper #3330: Help mini12864 LCD Display  voron 2.4

Alexander19831, 2020-09-15:
не могу разобрать PIN и он не работает ;-( 

[display]
##	mini12864 LCD Display
lcd_type: uc1701
cs_pin: z:P1.18
a0_pin: z:P0.15
rst_pin: z:P0.18
encoder_pins: ^z:P3.25,^z:P3.26
click_pin: ^!z:P0.28
contrast: 63

[neopixel fysetc_mini12864]
##	To control Neopixel RGB in mini12864 display
pin: z:P0.16
chain_count: 3
initial_RED: 0.4
initial_GREEN: 0.0
initial_BLUE: 1.0
color_order_GRB: False

##	Set RGB values on boot up for each Neopixel. 
##	Index 1 = display, Index 2 and 3 = Knob

[delayed_gcode setdisplayneopixel]
initial_duration: 1
gcode:
        SET_LED LED=fysetc_mini12864 RED=1 GREEN=1 BLUE=1 INDEX=1 
        SET_LED LED=fysetc_mini12864 RED=1 GREEN=0 BLUE=0 INDEX=2 
        SET_LED LED=fysetc_mini12864 RED=1 GREEN=0 BLUE=0 INDEX=3 TRANSMIT=0

ShohninDmitriy, 2020-09-15:
https://github.com/KevinOConnor/klipper/issues/2832#issuecomment-624772482

Alexander19831, 2020-09-15:
Спасибо вам больше!

ShohninDmitriy, 2020-09-15:
проблемма решилась ? Заработал ?

Alexander19831, 2020-09-15:
> проблемма решилась? Заработал?

Да, но в меню экрана octoprint и половина не работает. А у меня Repiter и как все это настраивается?

ShohninDmitriy, 2020-09-15:
Меню надо делать самому, оно сделано по дефолту простеньким описание в [конфигах]( https://github.com/KevinOConnor/klipper/blob/master/config/example-menu.cfg ) и в [файле пример дефолта](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/display/menu.cfg)

closed 2020-09-22 by klipper-gitissuebot