Klipper3d/klipper #3331: MCU 'mcu' shutdown: Timer too close

pandaym3D, 2020-09-15:
This issue has really been a major problem for me ever since I started using Klipper, and everytime I think I solved it, it just seems to happen again.

I am running Klipper on the Einsy Klipper of my Prusa MK3S. My Raspberry is a 3B+, and I am not using any plugins for octoprint or running anything else on it.

I have talked to a lot of people regarding the issue, and tried all sorts of fixes. Some of those being:
- Tried various USB cables (shorter/longer, with/without ferrite core etc.)
- Flashed the 32u2 bootloader from PrusaOwners
- Lowered my slicer resolution
- Only running 16 microsteps
- Put a heatsink on my X stepper and some foam heat shielding around the USB connector to avoid the heat from the X stepper influencing the connection

Being at my wits end about it, I bought a Duet WiFi to try and replace the Einsy Rambo with. Unfortunately though, I cannot try that solution until there is 2660 sensorless homing support in Klipper, or I have made an end-stop switch solution.

Any indications in my log as to why this is happening?

[klippy.log](https://github.com/KevinOConnor/klipper/files/5223254/klippy.log)

sergiocipo, 2020-09-15:
looking at you log your heatbed failed. heater_bed: target=60 temp=36.9 pwm=1.000. it means that you target temperature was  60 the actual temperature 36.9 and pwm 1 indicates that firmware was tring to het up your bed.

pandaym3D, 2020-09-15:
> looking at you log your heatbed failed. heater_bed: target=60 temp=36.9 pwm=1.000. it means that you target temperature was 60 the actual temperature 36.9 and pwm 1 indicates that firmware was tring to het up your bed.

Hmmm, that is very weird. I got no idea if you are right, as I cannot read the log, but I find it very hard to see that being the case. It would be a very weird coincidence if my heated bed suddenly had electrical/mechanical issues starting the very same day I started using Klipper.

I just noticed there was another case here where you concluded the same: https://github.com/KevinOConnor/klipper/issues/3316

Could it be how Klipper handles the heated bed?

sergiocipo, 2020-09-15:
Oh, just got too long into the log, the error was way before that line.

KevinOConnor, 2020-09-15:
The error occurred because the host machine became temporarily overloaded.  (The temperature control thread was unable to get any processing time for 450ms.)

It's unusual to see problems like this on an non-customized rpi3.

There was a recent report of a similar issue that may be related to the use of input_shaper in #3316 .  Does the problem go away if you disable input_shaper?  If so, can you provide the g-code file that causes the issue when input_shaper is on?

Otherwise, does the host machine have a webcam attached to it?  Is it possible the host machine is not getting a stable voltage (or is overheating)?

-Kevin

mdbro, 2020-09-15:
I encountered the same error on my last two prints.  I printed two parts, the print failed in the middle of the print with the error _'mcu' shutdown: Timer too close_.  I flipped one of the parts and tried printing again and this print also failed midway through the print with the same error.

I have pressure advance and input_shaper turned on and I recently increased my pressure_advance.

I've attached today's klippy.log, as well as the two relevant gcode files and my current printer.cfg:

[klippy.log](https://github.com/KevinOConnor/klipper/files/5228307/klippy.log)

[failed1.gcode.txt](https://github.com/KevinOConnor/klipper/files/5228308/failed1.gcode.txt)
[failed2.gcode.txt](https://github.com/KevinOConnor/klipper/files/5228309/failed2.gcode.txt)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5228303/printer.cfg.txt)

pandaym3D, 2020-09-15:
> The error occurred because the host machine became temporarily overloaded. (The temperature control thread was unable to get any processing time for 450ms.)
>
> It's unusual to see problems like this [...]

The RPi3 is definitely completely uncustomized.

I am not sure if the problem goes away when not using input shaper, and it is a little bit hard to test. I am currently doing ~12-2+ hour prints, and this last one failed at 92%. I was not able to figure out a way to continue the print either.

Here is the gcode I had issues with though:
[V3.0.39 Frame P1 (Part B)_0.15mm_PLA_MK3S_11h43m.txt](https://github.com/KevinOConnor/klipper/files/5228593/V3.0.39.Frame.P1.Part.B._0.15mm_PLA_MK3S_11h43m.txt)

There's no webcam or anything attached to the RPi, it is a really basic Klipper-OctoPrint install. Only plugin I run is the OctoKlipper one in fact. It runs on the original powersupply.
By host machine, I assume you mean the printer. That is running on the stock Prusa PSU. 

Regarding overheating, I did notice that my steppers do get hotter than I'm used to since I started using Klipper, but I am running a lot faster now too. So a theory I had was that as the X stepper, that gets fairly hot, passes by the Einsy control box as along the print, it eventually passes the USB cable connection on top of the Einsy box. I was thinking that the USB cable could become so hot that, well, it could cause problems(I don't technically know why). So therefore I slapped on a heatsink for steppers I had lying around on the X motor, and cabletied some foam on the USB connector to shield it from heat radiation. That however did not solve it.

As I had tried so many things already, I decided to go to a local electronics store and pick up a new RPi 4 4GB along with their original USB-C PSU and a fresh high quality micro SD. I reinstalled onto that, and I am now running the same print. Too early to say if it solved anything of course. But since you say it might be input_shaper related, I am thinking those algorhitms are handled on the Pi, and therefore more processing power here might theoretically help? 
Reason I did decide to grab a new Pi, was that the Pi3b+ was fairly old(from when they were released), and had seen a lot of use as a home automation server before, so even though I know very little about it, my thought was it may be somewhat worn.

Interesting that there seem to be more cases of this. Looking forward to hearing about what you may find in my GCODE etc.

dmbutyugin, 2020-09-15:
@pandaym3D I tried running your gcode in Klipper debug mode on my laptop (modulo bed meshing that I naturally cannot reproduce), and did not observe any suspicious stuckiness or anything like it (e.g. if the stepper kinematics code would get stuck somewhere). Note that Pi3b+ should be, in principle, sufficient for input shaper - I am running my delta printer with Pi3b+ and 2hump_ei input shaper, and both delta kinematics and this input shaper is more computationally demanding than mzv input shaper and cartesian kinematics.

KevinOConnor, 2020-09-16:
> By host machine, I assume you mean the printer. That is running on the stock Prusa PSU.

No, I mean the Raspberry Pi.  The error was in the host python code.  It's unlikely the micro-controller or steppers could cause the events indicated in the log (unless, indirectly, for example, by overheating / undervoltaging the rpi).

> therefore more processing power here might theoretically help?

It might help, but as @dmbutyugin indicates, there's plenty of power in an rpi3, so a bit of a mystery why it occurred.

-Kevin

pandaym3D, 2020-09-16:
Hmm that is indeed odd then.

I may add to the case that the reason I moved to Klipper was because I was intrigued by the resonance compensation features. Therefore, first thing I did after installing Klipper was tune my printer throughout, including input shaper. So I have barely run the printer without this enabled. If input shaper is indeed the source of the problem, and it can occur 10+ hours into a print, it is really hard to reproduce I guess.

Also I will note that I have done a couple of fairly long prints (~15-16 hours) without the issue. I have probably had it occur in 1/3rd of my prints for a total of maybe 10 times since I started using Klipper.

First 14 hour print succeeded without issues with the RPi 4 now, and started the next one right after. I guess the next couple of days will give an indication of whether the new Pi did anything.

puneit-hassio, 2020-09-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5231101/klippy.log)
@KevinOConnor 
I am facing the same issue - getting the error for prints longer than 5 hours ; around 3-4 hours from starting the print.  I am on raspberry pi zero W with latest Klipper, Moonraker and Mainsail

dmbutyugin, 2020-09-17:
So, I tried to add the measurement of time to toolhead.py for lookahead queue processing and _process_moves method and run the gcode from pandaym3D in Klipper batch mode on my RPi 3b to make it more representative of the real conditions, but to no avail. The processing time for either lookahead queue or _process_moves never exceeded 0.02 seconds, save exceeding 0.1 seconds. Though I had to use `-o /dev/null` to avoid writing large serial file to the sdcard, which can get slower. So, from this I am reasonably positive that input shaper code itself should not be a reason for this timer too close errors, at least in pandaym3D's case.

Separately, Kevin, I thought that the buffer in Klipper is around 2 seconds, so a ~450 msec scheduling delay of the processing thread is not ideal, but should it lead to an error like this?

KevinOConnor, 2020-09-17:
> So, from this I am reasonably positive that input shaper code itself should not be a reason for this timer too close errors, at least in pandaym3D's case.

FYI, the log in this issue did not have any indicators of correlation with input_shaper.  The log here shows the thread running the PID didn't get CPU time for 450ms, and the log doesn't show Klipper doing anything during that time.  In contrast, issue #3316 shows Klipper trying to generate steps when it eventually runs out of time.

> I thought that the buffer in Klipper is around 2 seconds, so a ~450 msec scheduling delay of the processing thread is not ideal, but should it lead to an error like this?

The log here (and in #3316) result in an error because the host PID code is unable to respond to a temperature measurement within 300ms.  (Specifically, the host reads a temperature which causes it to send a heater update, but that heater update message only gets sent to the micro-controller after its deadline has elapsed.)  So, yes, the kinematics have much more time, but the PID control loop was what exhibited the error.

In both cases, though, this should not have happened, as the PID runs in its own thread.  So, even if input_shaper was running for a very long time, it should not have stopped the PID from updating.  (The input_shaper C code does not block the Python PID code.)

FYI, I've been chasing down a possible interaction with the Python garbage collector, which could cause the PID thread to be delayed.  I don't know if this is the root cause of these recent issues, but I do see ways to improve the garbage collection to avoid interference.

-Kevin

Tahutipai, 2020-09-17:
@pandaym3D  Is it possible this was caused by undervolting of the Pi? I note you have the original power supply, but there [have been cases](https://www.element14.com/community/thread/64029/l/under-voltage-detection-raspberry-pi-3-b) of under volting with with the original power supply, for various reasons.

If you still have the old pi setup, perhas you can put it under load and check ``vcgencmd get_throttled``, or there is a handy python script at the end of (this page)[https://harlemsquirrel.github.io/shell/2019/01/05/monitoring-raspberry-pi-power-and-thermal-issues.html] to check undervolting.

Does your power supply have the cable directly attached to it, or is there a USB cable between your power supply and the Pi? If the latter, it's may be worth swapping out that cable for a known quality one.

KevinOConnor, 2020-09-17:
I've updated the code to improve the handling of "python garbage collection" (primarily commit cb0a8f2e).  Anyone that is getting "Timer too close" errors should update to the latest code (see the FAQ for upgrade instructions).  It is not clear if this will resolve the issue - if you continue to encounter these errors, please report them (along with the associated klipper log).

-Kevin

pandaym3D, 2020-09-17:
> @pandaym3D Is it possible this was caused by undervolting of the Pi? I note you have the original power supply, but there [have been [...]

I'm sorry but I don't know much about the power supplies etc. so I don't know. I have just run it with the original PSU, thinking that would be good.

Is there a smart way to put the Pi under load and then run vcgencmd get_throttled ? I'll quickly hook up the 3B+ and do that then.

puneit-hassio, 2020-09-19:
Update:
I moved from pi zero w to pi3 B+. Since, it was a fresh install, it got the latest commit. Since then I haven't faced the problem of timer too close. Do not know if its was the pi or the commit.

pandaym3D, 2020-09-21:
Update from me too. Since I moved to Raspberry Pi 4 4GB, I have not faced this issue. I have not updated to the new commit -admittedly because I did not know how.

If you guys wish me to test my Pi3 for undervolting, I can do that if you guide me a little about what to do :)

KevinOConnor, 2020-09-22:
The upgrade instructions are in the FAQ ( https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software ).  I don't see a reason to go back to an rpi3, unless you want to.  It is probably worthwhile to upgrade to the latest code (on the rpi4 and/or rpi3) as there is a chance the problem could occur even with an rpi4.

-Kevin

closed 2020-10-21 by github-actions[bot]

DroneMang, 2021-01-13:
Having the same issue today, just started yesterday after changing the bed mesh settings.
[klippy8.zip](https://github.com/KevinOConnor/klipper/files/5806082/klippy8.zip)