Klipper3d/klipper #3334: Z Offset with BLTouch does not work when printing

Twilek-de, 2020-09-15:
I have encountered a weird problem with the Z Offset. I have a CR 10 V3 with BLTouch. I have done the PROBE_CALIBRATION procedure multiple times and always came out with something around 3.5. Having set that lowering the nozzle to the bed manually works fine and I get a good grip on the paper on every part of the bed with the paper test. The moment I start printing the nozzle will fly about a millimeter above the bed and print in the air. After G28 and manually lowering everything is fine again. If I manually compensate for this and set the Z Offset to 4.4 I get a good first layer. This is 100% reproducible

[image]

[image]

This is the height of the nozzle while printing the purge line

[image]

This is the resulting purge line

[image]

This is the nozzle height after g28 and g1 z0

[image]

And this is the purge line after manually overcompensating the z offset to 4.4

[klippy.log](https://github.com/KevinOConnor/klipper/files/5228336/klippy.log)

Twilek-de, 2020-09-15:
Might be the same as this https://github.com/KevinOConnor/klipper/issues/3324 and then again maybe not....

CHILLYSMOKES, 2020-09-15:
Comment out the Z-Offset under the Extruder section. Then make sure you are using the offset in the BLTouch section. I had same issue until I did that. Now it works flawlessly.

Twilek-de, 2020-09-15:
@CHILLYSMOKES Which entry in my config are you refering to? I can only find a z_offset in the bltouch section. The extruder section has the following settings:
[extruder]
step_pin = ar26
dir_pin = !ar28
enable_pin = !ar24
step_distance = .002266032
nozzle_diameter = 0.400
filament_diameter = 1.750
pressure_advance = 0.12
pressure_advance_smooth_time = 0.04
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
min_temp = 0
max_temp = 300
control = pid
pid_kp = 24.967
pid_ki = 1.290
pid_kd = 120.780

CHILLYSMOKES, 2020-09-15:
I meant the stepper_z section... Duh. Sorry. Look in there.

Twilek-de, 2020-09-15:
Hmm still not sure ... that section looks like 
[stepper_z]
step_pin = ar46
dir_pin = !ar48
enable_pin = !ar62
step_distance = .0025
position_max = 400
position_min = -3
endstop_pin = probe:z_virtual_endstop

I have searched my whole config for the word "offset" and it only shows up in the bltouch section und in the start print macro (I have tried commenting it out there, but that didn´t help)

Gilabite, 2020-09-15:
The z offset is in the bltouch section. If you ran the probe calibration and saved it, it should have commented out the zoffset in the bltouch section and it puts it at the bottom in the do not edit below this block section of the config.

mental405, 2020-09-16:
I would delete the entirety of the section at the bottom of the config that says "DO NO EDIT BELOW THIS LINE".

Then run probe calibrate until you get an offset you like, then be sure to run SAVE_CONFIG

Also, if you happen to be using OctoKlipper, do not edit your config file after running SAVE_CONFIG unless you first close and re-open the OctoPrint browser window.

Twilek-de, 2020-09-16:
I am aware how the save config mechanic works and I had only one uncommented z_offset at any given time being it manually entered in the upper bltouch section or auto generated in the lower block with the upper one commented out. That had absolutely no effect on the differences in z offset while manually moving the head and printing.

CHILLYSMOKES, 2020-09-16:
Post your printer.cfg please. And check your slicer for Z-Offset.

Twilek-de, 2020-09-16:
I am using Cura without any z offset modification that I am aware of.

My printer.cfg looks like this  (commenting out the "SET_GCODE_OFFSET Z=0.0" doesn´t change the problem):

#CR-10 V2 PLUS (From Tiny Machines: CR-10 V2 with bl touch and titan direct drive included MAY VARY FROM STANDARD WIRING ON THE BLTOUCH)
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
position_max: 400
position_min: -3
endstop_pin: probe:z_virtual_endstop

[safe_z_home]
home_xy_position: 104.6,150.4 # Change coordinates to the center of your print bed relative to the probe
speed: 50
z_hop: 10
z_hop_speed: 5

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
step_distance: .002266032
nozzle_diameter: 0.400
filament_diameter: 1.750
pressure_advance: 0.12
pressure_advance_smooth_time: 0.04
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
#control: pid
#pid_Kp: 22.2
#pid_Ki: 1.08
#pid_Kd: 114
min_temp: 0
max_temp: 300

[heater_bed]
heater_pin: ar8
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
#control: pid
#pid_Kp: 690.34
#pid_Ki: 111.47
#pid_Kd: 1068.83
min_temp: 0
max_temp: 100

[... 272 lines cut ...]
#*# tension = 0.2
#*# min_x = 48.0
#*# algo = bicubic
#*# y_count = 7
#*# mesh_y_pps = 2
#*# min_y = 6.0
#*# x_count = 7
#*# max_y = 300.0
#*# mesh_x_pps = 2
#*# max_x = 300.0
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 24.967
#*# pid_ki = 1.290
#*# pid_kd = 120.780
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 72.505
#*# pid_ki = 1.465
#*# pid_kd = 897.251
#*#
#*# [bltouch]
#*# z_offset = 4.350

CHILLYSMOKES, 2020-09-16:
Twilek- I see multiple G28 in your start gcode. Did you manually change the very last line?

Twilek-de, 2020-09-16:
I cut the last lines straight out of cura, so that I am just using a call to the macro as cura start gcode. But it really can´t be the last line only because the nozzle purge line (the gcodes before that) already have the problem.

Twilek-de, 2020-09-17:
As there have been changes to the bed leveling code in the last two days I thought it might be a good idea to update the firmware and I did. It seems to work now, which I guess is good. But I am a bit cautious, as it also worked when I flashed the klipper fw for the first time. I hope it is not something that gets reset when flashing the firmware and then slowly builds up again. We will see I guess.

CHILLYSMOKES, 2020-09-17:
Close it then bro. There are other issues still open that are similar enough if they want to monitor. GJ.

Twilek-de, 2020-09-17:
I thought to give it a day or two to see wether it stays working. But you are right, I can close it and if necessary open it again.

Twilek-de, 2020-09-20:
And it´s back. After several succesfull prints, the nozzle is "air printing" again, as before....

jrgilman, 2020-09-20:
@Twilek-de I think I'm having a very similar issue or the same... 

My first layer is going down really poorly, specifically in one corner where the nozzle ends up hovering visibly above the build plate. It ends up just dragging the filament along for the ride until it catches. The second layer _seems_ to go down correctly though?

.I've spent 8 hours yesterday and several today trying to square my ender 3 to absolute perfection, along with adjusting the eccentric screw on the x gantry to level the bed as best as humanely possible (with the aid of the bed visualizer).

Problem Corner Image: https://cdn.discordapp.com/attachments/582347488938164240/757297428167655485/image0.jpg
GCode for the calibration print (in-case some slicer setting is interfering): 
`; G-Code generated by Simplify3D(R) Version 4.1.2
; Sep 20, 2020 at 12:37:58 PM
; Settings Summary
;   processName,Process1
;   applyToModels,i3_mega_bed_level
;   profileName,Ender 3 PLA - No Raft (modified)
;   profileVersion,2020-09-15 08:39:39
;   baseProfile,Wanhao Duplicator i3 (modified)
;   printMaterial,PLA
;   printQuality,Fast
;   printExtruders,
;   extruderName,Primary Extruder
;   extruderToolheadNumber,0
;   extruderDiameter,0.4
;   extruderAutoWidth,1
;   extruderWidth,0.48
;   extrusionMultiplier,0.8
;   extruderUseRetract,1
;   extruderRetractionDistance,5
;   extruderExtraRestartDistance,0
;   extruderRetractionZLift,0
;   extruderRetractionSpeed,2400
;   extruderUseCoasting,0
;   extruderCoastingDistance,0.06
;   extruderUseWipe,0
;   extruderWipeDistance,3
;   primaryExtruder,0
;   layerHeight,0.16
;   topSolidLayers,8
;   bottomSolidLayers,7
;   perimeterOutlines,3
;   printPerimetersInsideOut,1
;   startPointOption,3
;   startPointOriginX,0
;   startPointOriginY,235
;   sequentialIslands,0
;   spiralVaseMode,0
;   firstLayerHeightPercentage,125
;   firstLayerWidthPercentage,100
;   firstLayerUnderspeed,0.3
;   useRaft,0
;   raftExtruder,0
;   raftTopLayers,3
;   raftBaseLayers,2
;   raftOffset,3
;   raftSeparationDistance,0.14
;   raftTopInfill,100
;   aboveRaftSpeedMultiplier,0.3
;   useSkirt,1
;   skirtExtruder,0
;   skirtLayers,1
;   skirtOutlines,2
;   skirtOffset,4
;   usePrimePillar,0
;   primePillarExtruder,999
;   primePillarWidth,12
;   primePillarLocation,7
;   primePillarSpeedMultiplier,1
;   useOozeShield,0
;   oozeShieldExtruder,999
;   oozeShieldOffset,2
;   oozeShieldOutlines,1
[... 3201 lines cut ...]
G1 X199.070 Y51.368 E15.8519
G1 X200.868 Y49.570 E15.9180
G1 X200.868 Y50.248 E15.9357
G1 X199.748 Y51.368 E15.9768
G1 X200.427 Y51.368 E15.9945
G1 X200.868 Y50.927 E16.0107
G92 E0.0000
G1 E-5.0000 F2400
; layer end
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G0 F2000 Z245 ; get gantry out of the way first
G0 F10000 Y235 ; move print out front
M84 ; disable motors
M106 S0 ; turn off cooling fan
; EndBeep
; M300 S294 P200 ;D4: 294
M117 Jobs Finished!
; Build Summary
;   Build time: 0 hours 33 minutes
;   Filament length: 795.7 mm (0.80 m)
;   Plastic volume: 1881.31 mm^3 (1.88 cc)
;   Plastic weight: 2.35 g (0.01 lb)
;   Material cost: 0.04
`

CHILLYSMOKES, 2020-09-20:
Have you guys tried using bed level compensation instead of mesh?

Twilek-de, 2020-09-20:
Nope, but I don´t think that that is the problem. When I lower the nozzle via manual controls to 0 height it will be exactly above the build plate at any position as it should. With exactly the same settings it will hover like 2 mm above the plate when trying to go to 0.2 for putting down the first layer. If I go to the same position afterwards and manually go to 0.2 the height is perfect again. And the strangest thing is. The problem went away and came back without me changing any parameters.

jrgilman, 2020-09-20:
> Have you guys tried using bed level compensation instead of mesh?

As in bed_tilt?

jrgilman, 2020-09-20:
@Twilek-de out of desperation I tried switching my algorithm from the default (lagrange) to bicubic in the [bed_mesh] section and I immediately saw a positive improvement in first layer adhesion. Maybe give it a shot?

Twilek-de, 2020-09-20:
> 
> 
> @Twilek-de out of desperation I tried switching my algorithm from the default (lagrange) to bicubic in the [bed_mesh] section and I immediately saw a positive improvement in first layer adhesion. Maybe give it a shot?

I am running in bicubic mode and again I really don´t think that it is a bed leveling problem. After a bed leveling manual or manually entered gcode driven positioning of the head works perfectly. Start a print and it all goes down the drain. When I have compensated the error the prints will churn out perfectly again. Somehow the software assumes a different Z Offset when printing than when idle. When I enter my reproducibly measurable Z Offset of 3.5 manual positioning will work every time at every position of the bed. I start a print and it adds an extra offset that sometimes changes. I haven´t found out when.

Twilek-de, 2020-09-20:
What I haven´t tried yet is what happens when I do a manual positioning after having initiated a bed leveling from the terminal. Until now i have only done manual positioning after a finished print, where the end print macro clears the mesh and cancelled prints with a firmware restart (which shouldn´t clear the mesh, right?!). I will try that when my current print finishes tomorrow and report back.

Twilek-de, 2020-09-21:
Ok I have been able to manually reproduce the problem. If I do a BED_MESH_CALIBRATE and the load the profile I will get the flying nozzle even with manual positioning. A PROBE_CALIBRATE will consistently give out a wrong parameter of 3.5 even after Mesh Calibration.

Twilek-de, 2020-09-21:
Ok something is indeed off with the bed leveling. I did a manual bed leveling. Then I did a mesh calibrate and saved that. Then I did a probe calibrate (which came up 3.5 again) and saved that. When I now go to Z=0 the nozzle is about 2mm above the bed on every position.

Twilek-de, 2020-09-22:
And suddenly it changed. I have done the same print twice directly after one another. The first print went perfectly. After having taken off the modells without any problems or force I started the second print without really checking how the first layer looked, as it had worked just 2 hours ago. This time the print was "rammed" into the build plate. I had to compensate about 0.2 mm afterwards and it then printed fine again.

jrgilman, 2020-09-22:
@Twilek-de I ended up figuring out what it was 100%. It had nothing to do with the algorithm and took me all weekend to identify. I have an Ender 3 so very similar to your CR-10. On the right-side of the x-gantry, the nylock nut holding the inner wheel was spinning freely and no longer was holding the wheel against the eccentric nut. I only noticed this when my hand slipped while messing around with the eccentric nut. Once I tightened this down via a ratchet + an allen wrench in the through hole on the other side my problems immediately went away. I would check your nylocks for something similar, just try turning them. They should be snug against the bearings...

Twilek-de, 2020-09-23:
Wow thanks for the effort with the photo and  it is really cool that you have your printer running again. For me I am afraid that wasn´t the solution. All screws and nuts of the X axis were nice and tight.

Twilek-de, 2020-09-24:
I have just disabled Klipper and moved back to Marlin as the Z Offset was all over the place. If the problem persists with Marlin I have to seriously take a look at my hardware. If not, something in Klipper is badly broken. I will report back.

Twilek-de, 2020-09-26:
I have done several prints with Marlin and haven´t seen any Z Offset problems, so it seems that something in Klipper didn´t work. As I have no further ideas how to solve the issue I will stay with Marlin for now...

closed 2020-09-26 by Twilek-de

Arksine, 2020-10-08:
@Twilek-de I just took a look at your issue and I notice that you are using a relative_reference_index in bed_mesh.  This will always override your z_offset when creating a mesh.  This setting should only be used if you are homing z with an endstop, as it override's the z-offset applied to the mesh.

Twilek-de, 2020-10-09:
I had to remove another g28 from my gcode which would disable the bed leveling. Then I reset the z offset and after some time the nozzle started flying again :-(.

jayzebra, 2021-08-10:
Hi,
I'm having this same issue on an Ender 3 Pro with a bltouch v3.
By now, I calibrated everything according to the instructions several times and I can't get the nozzle to stay close to the bed. I even tried several values from 0 to 2 for the z_offset . The nozzle just hoovers around 1 to 2mm above the bed with the exact same issue the original poster described.

# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 30

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 30

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
position_max: 250
position_min: -1

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
microsteps: 16
rotation_distance: 31.917
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
min_temp: 0
max_temp: 255
#pressure_advance: 0.14

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

[bed_screws]
screw1: 50,50
screw2: 200,50
screw3: 50,200
screw4: 200,200

# BLTouch probe. One may define this section (instead of a probe
# section) to enable a BLTouch probe. (Note! This bltouch module may
# not work correctly with some BLTouch "clones"!) A virtual
# "probe:z_virtual_endstop" pin is also created (see the "probe"
# section above for the details).
[bltouch]
sensor_pin: ^PB7
control_pin: PB5
#pin_up_touch_mode_reports_triggered: False
#pin_up_reports_not_triggered: False
#sensor_pin: ^PC4
#   Pin connected to the BLTouch sensor pin. This parameter must be
#   provided.
#control_pin: PA4
#   Pin connected to the BLTouch control pin. This parameter must be
#   provided.
#pin_move_time: 0.675
#   The amount of time (in seconds) to wait for the BLTouch pin to
#   move up or down. The default is 0.675 seconds.
#pin_up_reports_not_triggered: True
#   Set if the BLTouch consistently reports the probe in a "not
#   triggered" state after a successful "pin_up" command. This should
#   be True for a genuine BLTouch; some BLTouch clones may require
#   False.  The default is True.
#pin_up_touch_mode_reports_triggered: True
#   Set if the BLTouch consistently reports a "triggered" state after
#   the commands "pin_up" followed by "touch_mode". This should be
#   True for a genuine BLTouch; some BLTouch clones may require
#   False. The default is True.

# Measure the x and y distance between the tip of your nozzle and 
# your BLTouch tip
# x : + is to the right and - is to the left
# y : + is toward the back and - is toward the front
x_offset: -42 # 41mm left of the nozzle
y_offset: -9 # 13mm forward of the nozzle
#z_offset: 0.400

speed: 5.0
samples: 2
sample_retract_dist: 8.0
set_output_mode: 5V
probe_with_touch_mode: True 

[safe_z_home]
home_xy_position: 158,125 # Change coordinates to the center of your print bed
speed: 20
#z_hop: 10                 # Move up 10mm
z_hop_speed: 5

[bed_mesh]
speed: 80
horizontal_move_z: 5
probe_count: 5,5
mesh_min: 50,30
mesh_max: 179,200

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 23.239
#*# pid_ki = 1.201
#*# pid_kd = 112.419
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 74.536
#*# pid_ki = 1.656
#*# pid_kd = 838.534
#*#
#*# [bltouch]
#*# z_offset =  0.101

This is my