Klipper3d/klipper #3337: Display Off at Shutdown

TheRunnerPi, 2020-09-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5232220/klippy.log)

Hi,

is it possible to turn off the display when the printer is in standby mode? 
When I switch to standby mode via OctoPrint everything is off, only the display is still lit.

Display is a "12864" and it is hanging on a ramps.
When I switch on the printer, the display is off for a few seconds before it starts up.  So it should be possible to turn off the display in standby mode by command, right?

Maybe someone has already solved something like this and can help me.

Thanks for your help.

KevinOConnor, 2020-09-22:
The "st7920" displays do not have the ability to turn off the backlight from software.

-Kevin

TheRunnerPi, 2020-09-22:
Okay, thanks for the Infomation. 
Then i will go for a second relay board to cut the power off.

Regards,
Daniel

closed 2020-09-22 by TheRunnerPi

homasholz, 2021-01-26:
is it possible on hd44780 ?