Klipper3d/klipper #3339: delta_calibrate on rotary delta printers does not work properly

klHaribo, 2020-09-16:
Dear Kevin,

I build the same rotary delta printer like vladT0 build (issue #1610).
but I am unable to calibrate him. I changed/improved every part, but I always end in a undesired concave or convex print bed.
playing with tuned values like shoulder_radius or shoulder_height only results into another z-height.
I measured carefully everything. But it is very difficult to find out the correct step_distance. 
So tried out the best value to reach z=0.
Is it possible to calculate the step_distance and position_endstop by measuring the height of the elbow homed and on z=0?
maybe these values will help to calculate a better calibration?

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5233323/klippy.log.zip)

-Dennis

klHaribo, 2020-09-17:
At my last try, I used the step_distance from vladT0 because it is the same setup.
i did a delta_analyze manual_height=225 
and a delta_calibrate method=manual
then I made a bed_mesh_calibrate. the printer was still convex, but I was able to print the calibrate_size (very ugly first layer due to manual filament feeding)
Finally I measured distances and got this:

[klippy.log](https://github.com/KevinOConnor/klipper/files/5239537/klippy.log)

DELTA_ANALYZE CENTER_DISTS=74.15,73.82,74.49,74.13,73.75,74.88
DELTA_ANALYZE OUTER_DISTS=74.43,74.55,73.60,74.65,73.85,73.85
DELTA_ANALYZE CENTER_PILLAR_WIDTHS=9.01,9.05,8.78
DELTA_ANALYZE OUTER_PILLAR_WIDTHS=8.91,8.94,8.66,8.84,8.90,8.70
DELTA_ANALYZE SCALE=1.0
DELTA_ANALYZE CALIBRATE=extended

Recv: // Internal error on command:"DELTA_ANALYZE"
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown

KevinOConnor, 2020-09-18:
> Recv: // Internal error on command:"DELTA_ANALYZE"

I fixed the above error (commit ad0b76bd).

However, it doesn't produce a useful result for that particular attempt - it doesn't look like the parameters from the last DELTA_CALIBRATE session matched the parameters used during the print.  Also, I don't think it would be valid to print the calibration object with bed_mesh enabled (the calibration code can't take into account the bed_mesh translations and thus wont calculate a good result).

> But it is very difficult to find out the correct step_distance.
So tried out the best value to reach z=0.
Is it possible to calculate the step_distance and position_endstop by measuring the height of the elbow homed and on z=0?

It's possible to "solve for step_distance", but that doesn't sound like it should be needed, as it should be possible to calculate the step_distance from the number of teeth on each pulley.

Maybe @vladT0 or @A1exKorvin have some advice on methods to calibrate and/or auto-calibrate.

-Kevin

vladT0, 2020-09-18:
Hi klHaribo!
I did not try it with latest release from Kevin. Printer is tuned on and working pretty good. I will try to find time at the weekend and recalibrate it with latest release step by step and let you know about results and publish log file. May be it will help.
Talk to you soon.

klHaribo, 2020-09-19:
oh yeah! that will be great!

I tried the last version. delta_analyze calibrate=extended worked, but it came to an error:
!! Endstop b still triggered after retract  
Mainsail said, endstop b was open...
Now I'm starting with a cleaned config again.

klHaribo, 2020-09-19:
OK, after this commit delta_calibrate seems to work fine for me.
no need for bed_mesh_calibrate ;)

vladT0, 2020-09-19:
If not urgent anymore I will not do it at the weekend :)

klHaribo, 2020-09-19:
No its not nessesary. After this change printer is working now. tunung will now be the easy part :)

closed 2020-09-19 by klHaribo