Klipper3d/klipper #3340: Failed to home probe: Timeout during homing

gomj9938, 2020-09-17:
I get this every couple of times doing G29. It usually just crashes into the build plate.

[printer.zip](https://github.com/KevinOConnor/klipper/files/5241567/printer.zip)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5241570/klippy.log)

KevinOConnor, 2020-09-22:
A "timeout during homing" occurs when the software does not receive a trigger signal from the endstop during the homing process.  Make sure to run the checks described in docs/Config_checks.md .  If your issue is with the bltouch, then the best advice we have is in the docs/BLTouch.md document.

-Kevin

gomj9938, 2020-09-23:
I had set 5v mode for the bltouch in Marlin a long time ago but had forgotten.  I just checked to make sure.  I set the bltouch to 5v and will try it for a couple of days but im mostly sure thats the problem.

closed 2020-09-24 by gomj9938