Klipper3d/klipper #3341: BLTouch on SKR 1.4Turbo

JuniorGamingTime, 2020-09-18:
I'm trying to setup my Hypercube that has a SKR 1.4Turbo with a BLTouch.
 bltouch is only showing Open no matter what. The pin is in the Z min connector and the servo in the servo slot. I was using marlin before with no issues.
 This is my first go with Klipper so im extremely confused.

My Config:
https://pastebin.com/qZxbEhWU

The log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5242893/klippy.log)

BlackStump, 2020-09-18:
I have no experience with that board but have you tried ^!P1.27
 >Pin
 names may be preceded by an '!' to indicate that a reverse polarity
 should be used (eg, trigger on low instead of high). Input pins may
 be preceded by a '^' to indicate that a hardware pull-up resistor
 should be enabled for the pin. If the micro-controller supports
 pull-down resistors then an input pin may alternatively be preceded
 by a '~'.

JuniorGamingTime, 2020-09-18:
I tried with and without the "!" and it just charges to Triggered with the "!" 
I haven't gotten it to be open one way and Triggered the other

Gomer05, 2020-09-18:
I'm just about to set up this board and my notes are following the BTT Servos and Probe docs, they say to use 0.10 for the probe.  Sometimes the endstop pins have pull-ups or pull-downs on them.  Any reason to not try the recommended pin?

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf

JuniorGamingTime, 2020-09-18:
I have the Turbo version but I could try the probe port tomorrow

BlackStump, 2020-09-18:
Pease read exactly what I suggested which is to include ^ symbol ie as I wrote      ^!P1.27 
I agree with @Gomer05  if you have a probe port on the board you should use that before using a endstop port.

Sineos, 2020-09-18:
I have a BLTouch working on SKR1.4 Turbo:

```
[bltouch]
#sensor_pin: P0.10
sensor_pin: P1.27
control_pin: P2.0
pin_move_time: 0.680
stow_on_each_sample: True
probe_with_touch_mode: False
```

For whatever reason the dedicated P0.10 didn't work for me and I gave up trying.

JuniorGamingTime, 2020-09-18:
@BlackStump i did what said, what i mean is i tried with and without the "!" so that mean "^!P1.27" and "^P1.27"

@Sineos i tried your config with no luck, the bltouch was working perfectly in marlin, even in marlin the probe port was an issue

Gilabite, 2020-09-18:
Not sure if it is the same on the turbo but I have the E3 DIP and the schematic for the bltouch port is wrong the signal and 5 volt lines are actually swapped from the schematic.

JuniorGamingTime, 2020-09-18:
I don't think so  on my other printer I've used the Probe port but I just prefer the Z min endstop.

closed 2020-10-17 by github-actions[bot]