Klipper3d/klipper #3342: Two questions about Neopixel

gitpower2017, 2020-09-18:
Why was the chain_count limited to a maximum of 18?
For example, I have 19 Neopixel LEDs.

Why is the brightness value defined between 1.0 and 0.0.
Wouldn't a value between 0 and 255 make more sense. 
(consistent with "normal" LEDs)

In an example for "normal" LEDs this is solved like this.
https://www.reddit.com/r/klippers/comments/cjp45u/temperature_controlled_leds/

[output_pin my_LED_pin_GREEN]
...
shutdown_value: 0
scale: 255

KevinOConnor, 2020-09-22:
> Why was the chain_count limited to a maximum of 18?

Klipper has an internal limit on message size (64 bytes) and the current low-level neopixel update message doesn't fit if there are more than 18 leds.

> Wouldn't a value between 0 and 255 make more sense.

No, I don't think there is anything intuitive about 255.  It is possible to write a gcode_macro that will convert from a scale of your choice - see docs/Command_Templates.md if interested.

-Kevin

gitpower2017, 2020-09-23:
Thanks for the Info.

There will be an adjustment for more than 18 neopixel LEDs

Gruß
Lothar

KevinOConnor, 2020-10-07:
FYI, there is a test branch with support for longer neopixel chains at #3380 .

-Kevin

altvnk, 2020-10-08:
My 2 cents on this: having color definition in form of 0-255 is not intuitive, but rather common for defining RGB colours. If you open any color palette, you will be able to get the RGB colour code to carry over. However, I do see a logic in current implementation, it perfectly maps on percentage, it's just not super convenient to use.

closed 2020-11-06 by github-actions[bot]