Klipper3d/klipper #3343: [FEATURE REQUEST] Add support for SET_COOLER_TEMPERATURE

andreashb2, 2020-09-18:
I have set up temperature_fan as following:

[temperature_fan chamber_fan]
pin: ar7
sensor_type: ATC Semitec 104GT-2 
sensor_pin: analog15
min_temp: 0
max_temp: 100
target_temp: 70
control: watermark
max_speed: 1.0
gcode_id: C0

I would like to control the target_temp using gcode commands such as you can do with a heater (SET_HEATER_TEMPERATURE HEATER=<heater_name> [TARGET=<target_temperature>]). That way you can control the target_temp for different materials, and also cool the chamber once the print is finished without editing the config file.

KevinOConnor, 2020-09-22:
There is currently no ability to change that target at runtime.  An interested developer would need to take this on and submit the code with the support.

-Kevin

KevinOConnor, 2020-09-23:
Sorry, there does exist a `SET_TEMPERATURE_FAN_TARGET` command - see docs/G-Codes.md for the details.

-Kevin

andreashb2, 2020-09-27:
Solved thanks!

closed 2020-09-27 by andreashb2