Klipper3d/klipper #3344: Help Request

bruce356, 2020-09-18:
Have a Cartesian IDEX printer with Klipper and Octopi.
Wish to setup a pause option to change filament and send T0 & T1 to their respective park position without cancelling the pause.

Depending on which tool is being used for printing the park positions are different T0 = X-41 & T1 = X291.

How to tell the tool being used to park in its appropriate X axis park position.
Thanks anyone that might be able to figure this out.
Bruce

bruce356, 2020-09-20:
This request for help would not require a log file as there is no fault to report, just need  help in creating a pause script for an IDEX printer as described above.
thanks

bruce356, 2020-09-20:
This is what I already have:-
After print job is paused.
{% if pause_position.x is not none %}
; relative XYZE
M117 Print Paused
G91
M83

; move Z 5mm upwards and retract filament by 2.0 mm. 
G1 Z+5 E-2.0 F4500

{% if pause_temperature[0] and pause_temperature[0]['target'] is not none %}
    M109 S{{ pause_temperature[0]['target'] }}
{% else %}
    M109 T{{ pause_temperature[0]['target'] }}
{% endif %}

; absolute XYZE
M82
G90

; move to a safe rest position, adjust as necessary (would like to be able to park X carriage in home position).
G1 X120 Y10 F8000
{% endif %}

KevinOConnor, 2020-09-20:
The config/sample-idex.cfg has example macros showing how to park the appropriate toolhead.  (In short, `PARK_{printer.toolhead.extruder}`.)

-Kevin

bruce356, 2020-09-22:
Hi Kevin, thanks for the reply, I have those macros in my printer config file, how would I instruct only the active used tool to park, somehow the gcode instruction needs to know which tool is actively being used and to send that tool only to the park position.
If both are simply included in the last instruction this will not work, for example:-

; move to a safe rest position, adjust as necessary
T0
T1
{% endif %}

or

; move to a safe rest position, adjust as necessary
PARK_extruder
PARK_extruder1
{% endif %}

Sorry I do not understand how I could write this last part of the script to perform what is needed.
Someone may have already done this.
Thanks - Bruce

ld3300, 2020-09-24:
If you include your log we can see the macros you have set up, but what I would do is something like the following:

When T0 is called
  park extruder1 with a PARK_extruder1 type macro
  ACTIVATE_EXTRUDER EXTRUDER=extruder

When T1 is called
  park extruder macro
  ACTIVATE_EXTRUDER EXTRUDER=extruder1

You may also want to use variables to track if something is already parked, but it may not matter depending on your setup

bruce356, 2020-09-25:
Hi ld3300, scripts for pause / resume on single tool printers are available and work well. 

I have attached my log file, not sure how helpful it would be. The pause/resume scripts that I am using are contained within Octopi under:-
"After print job is paused" 
{% if pause_position.x is not none %}
; relative XYZE
M117 Print Paused
G91
M83
; move Z 5mm upwards and retract filament by 2.0 mm. 
G1 Z+5 E-2.0 F4500
; Lower tool temperature if needed
M104 S105
; absolute XYZE
M82
G90
; move to a safe rest position, adjust as necessary
G1 X120 Y10 F8000
{% endif %}
;---------------------------------------
; With the "G1 X120 Y10 F8000" parmeter this works perfectly well because it is a location that either tool0 or tool1 (depending on which one is being actively used) can access.
Needed is a command that on pause, would park the active tool to its home/park position automatically.
The script or command needs to know which tool is active in order to park that tool and keep that tool active for resume.

If something simple like T0, T1, is called, which ever of those two is called last in the script, is the tool that remains active, so on resume that last called/selected tool will be the one that continues with the print, but it may be the wrong tool.
;----------------------------------------
"Before print job is resumed"
{% if pause_position.x is not none %}
; relative extruder
M117 Print Resumed
M83
; add this to the resume script to reload the temperatures as at time of pause.
{% for tool in range(printer_profile.extruder.count) %}
    {% if pause_temperature[tool] and pause_temperature[tool]['target'] is not none %}
        {% if tool == 0 and printer_profile.extruder.count == 1 %}
            M109 S{{ pause_temperature[tool]['target'] }}
        {% else %}
            M109 T{{ tool }} S{{ pause_temperature[tool]['target'] }}
        {% endif %}
   {% else %}
        {% if tool == 0 and printer_profile.extruder.count == 1 %}
           M104 T{{ tool }} S0
       {% else %}
           M104 S0
       {% endif %}
    {% endif %}
{% endfor %}
; prime nozzle
G1 E-0.8 F4500
;G1 E1.1 F4500
; G1 E0.8 F4500
; absolute E
M82
; absolute XYZ
G90
; reset E
G92 E{{ pause_position.e }}
; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F10000
; Set the Speed to something reasonable.
G1 F2000
; reset to feed rate before pause if available (THIS DOES NOT WORK)
;{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
;{% endif %}
;-----------------------------------------

These scripts are not my creation (very limited understanding), they have been copied with some changes to parameters only.
Regards

[klippy.zip](https://github.com/KevinOConnor/klipper/files/5280675/klippy.zip)

ld3300, 2020-09-26:
Hi @bruce356,

I have submitted #3368 as an example for a dual extruder single hot end.  Even though not exactly the same as your setup some of the principles apply.  You can also do something like Kevin recommends where you can do something like:

```
[gcode_macro T0]
gcode:
  {% if (printer.toolhead.extruder) == extruder1 %}
    ---insert extruder1 park position here---
  {% endif %}
  ACTIVATE_EXTRUDER EXTRUDER=extruder

[gcode_macro T1]
gcode:
  {% if (printer.toolhead.extruder) == extruder %}
    ---insert extruder park position here---
  {% endif %}
  ACTIVATE_EXTRUDER EXTRUDER=extruder1
```

I'll comment your above code for you as much as possible if it helps remove redundancy and simplify your needs. 

-David

bruce356, 2020-09-29:
Hi David @ld3300 , very much appreciate your help, sorry I am just not able to make what I would like to work.
Printing parts is easy compared to being able to create functioning scripts.

I added your script above to the T0 & T1, have tried so many variations (due to lack of knowledge) can not remember everything. This should be simple but it is not, have looked at your change filament scripts, its not simple for me.

Tried moving the pause/resume scripts from Octopi to printer.cfg, the pause script is all good but the resume script came up with syntax error, "expected : but got }" no clue how to fix this.

I am just going to have to be satisfied with parking at a common location that both tools can reach "G1 X120 Y10".
Regards - Bruce

closed 2020-10-01 by bruce356