Klipper3d/klipper #3345: Sensorless homing with tmc2209. Multiple g28s in a gcode file only homes the first time

bondus, 2020-09-18:
Having a gcode file with multiple G28 with movement in between only the first G28 works properly. The following G28s trigger the sensorless endstops immediately and do not move the axis at all.

Adding a TMC_SET_CURRENT in [homing_override] works around the issue and it works as intended.

Running two G28 after each other on the console works good, it's just when it's in a file on the virtual SD it does not work properly. 

Two G28 in the same g-code file works if there are no movements between them.

If I'm allowed to guess some registers in the TMC is not set/reset properly before homing and there is a lingering stallguard trigger from the previous moves.

[klippy_log_working.txt](https://github.com/KevinOConnor/klipper/files/5247256/klippy_log_working.txt)
[klippy_log_not_working.txt](https://github.com/KevinOConnor/klipper/files/5247257/klippy_log_not_working.txt)

Sorry for the extra debug printouts in the logs, I'm working on a new kinematics. There is not much interesting in them anyway.

bondus, 2020-09-19:
It was missing the start of the log with versions and configs:
[klippy_start_log.txt](https://github.com/KevinOConnor/klipper/files/5248564/klippy_start_log.txt)

KevinOConnor, 2020-09-22:
As far as I know, this quirkiness is due to the trinamic drivers themselves.

Since you're comfortable with the code, you can add debugging to klippy/extras/tmc.py:TMCVirtualPinHelper if you think the code is doing something wrong.

-Kevin

Ducati87, 2020-09-30:
Same problem. How do you solve it?

Dzikibiszkopt, 2020-09-30:
In my corexy machine I use G4(dwell) command between homing x/y axes to give time for tmc drivers to reset thier state on diag output because both of them will get trigger after stopping on one of axis.
Try G4 P2000

Ducati87, 2020-09-30:
Do you mean:
G28 X
G4 P2000
G28 Y
Right?

Dzikibiszkopt, 2020-10-01:
Exactly

bondus, 2020-10-28:
"github-actions" is getting impatient. I'm about to set up the machine that had these issues again. 
Once reproducible in a proper environment it's hopefully easy to track down the source of this little issue, and fix it.

closed 2020-11-27 by github-actions[bot]