Klipper3d/klipper #3347: Klipper not installing at all

Fabrizio3, 2020-09-19:
no klippy.log as the problem is in the instalation
(im running this and have been perfectly for 2 years on a linux mint pc till today when i wanted o update klipper)

so, after updaring klipper failed and threw out errors, i decided to reinstall it as per the similar threads suggested, i uninstalled everything according to your guide, deleted all folders and went to copy the repository in and run the ./klipper/scripts/install-octopi.sh (also tried debian and ubuntu) and it doesn't install, it doesn't even go into trying to install, just does sudo apt-get and stops right after, the klipper service isn't installed as it tells me that when i would run klipper service start, i really don't know how to make it work since i just barely managed to scrape by 2 years ago with a lot of headache and hours of troubleshooting

seth-cohen, 2020-09-19:
what is your default version of Python? I think klipper may still require python2 - is that available on your system.

Any errors written to stderr when running the script?

Fabrizio3, 2020-09-19:
iirc it's running python 2.7, haven't touched anything since I installed klipper almost two years ago, I'd have to check that stderr, let me see what it is and where it's at lol, not a linux user

Fabrizio3, 2020-09-19:
don't know where to look at that, but by taking a peek in the install script, I see it's failing either in the sudo apt-get which throws some errors about not having the public key (Spotify and chrome) or the sudo apt-gey install, doesn't install so my guess is at the update

Fabrizio3, 2020-09-19:
SOLVED: I commented outthe "force script to exit if error occurs" ( "set -e) and it worked, spotify and chrome couldnt get apt-get update so it crashed the whole script, while running it by hand i noticed that and gave it a go, klipper appears to have installed, will update soon

closed 2020-09-19 by Fabrizio3