Klipper3d/klipper #3348: Failed automated reset of MCU 'mcu' after FIRMWARE_RESTART on Creality V4.2.2

jasonttech, 2020-09-19:
Any time I do a FIRMWARE_RESTART or something that causes a disconnect like trying to move before homing axis, I cannot get it reconnected except by disconnecting USB, power cycling the printer, and restarting the klipper service.  Everything else seems to be working correctly. Below are the details of the printer, board, and menuconfig.

Printer: Ender 3 Pro
Board: Creality V4.2.2 (this is the same board used in the Ender 3 V2)

 ```
/home/pi/klipper/.config - Klipper Firmware Configuration
 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
  lqqqqqqqqqqqqqqqqqqqqqqq Klipper Firmware Configuration qqqqqqqqqqqqqqqqqqqqqqqqk
  x  Arrow keys navigate the menu.  <Enter> selects submenus ---> (or empty       x
  x  submenus ----).  Highlighted letters are hotkeys.  Pressing <Y> includes,    x
  x  <N> excludes, <M> modularizes features.  Press <Esc><Esc> to exit, <?> for   x
  x  Help, </> for Search.  Legend: [*] built-in  [ ] excluded  <M> module  < >   x
  x lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk x
  x x    [*] Enable extra low-level configuration options                       x x
  x x        Micro-controller Architecture (STMicroelectronics STM32)  --->     x x
  x x        Processor model (STM32F103)  --->                                  x x
  x x        Bootloader offset (28KiB bootloader)  --->                         x x
  x x        Clock Reference (8 MHz crystal)  --->                              x x
  x x    [ ] Use USB for communication (instead of serial)                      x x
  x x    [ ] Use CAN for communication (instead of serial)                      x x
  x x        Serial Port (USART1)  --->                                         x x
  x x    (250000) Baud rate for serial port                                     x x
  x x    [ ] Specify a custom step pulse duration                               x x
  x x    ()  GPIO pins to set at micro-controller startup                       x x
  x x                                                                           x x
  x x                                                                           x x
  x x                                                                           x x
  x x                                                                           x x
  x x                                                                           x x
  x x                                                                           x x
  x mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj x
  tqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqu
  x                     <Select>    < Exit >    < Help >                          x
  mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

```

[klippy20200918-2025.log](https://github.com/KevinOConnor/klipper/files/5248816/klippy20200918-2025.log)

pcdiks, 2020-09-19:
I have seen the same. I only use the restart command, that works for me.

jasonttech, 2020-09-19:
> I have seen the same. I only use the restart command, that works for me.

That works sometimes for config changes, but if I do anything that causes a disconnect like a move without home the RESTART doesn't clear it and I need to power cycle.

jasonttech, 2020-09-19:
I ending up needing to add `restart_method: command` to the mcu config to fix this. Did PR #3349 to get this in the example configs for the printers that use this board.

jasonttech, 2020-09-22:
commit 08eec0d

closed 2020-09-22 by jasonttech