Klipper3d/klipper #3352: Y Stepper driver shuts down mid print

paul-arnold, 2020-09-20:
Hi folks,

I noticed that my Y stepper driver had a shut down mid-printing again.
This also occured a few days ago but I thought it was a one-time thing as it never happened before.
So I restarted the same gcode and everything went fine.

Now it happened again. 
All drivers (tmc2208) are actively cooled and are being controlled via uart. Unfortunately I cant really see something myself in the klippy log, so I must rely on your help in this matter.
Thanks in advance

[klippy.log](https://github.com/KevinOConnor/klipper/files/5252430/klippy.log)

Roycinger, 2020-09-21:
Update: 
Y driver shut down again 4 hours into a print.
It's cool to the touch, DUMP_TMC gave me this:
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 0008190f IHOLD=15 IRUN=25 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000013 TPWMTHRS=19
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000003 reset=1 drv_err=1(ErrorShutdown!)
Recv: // IFCNT:      0000000c IFCNT=12
Recv: // OTP_READ:   0000000d OTP_FCLKTRIM=13
Recv: // IOIN@TMC220x: 20000341 ENN=1 PDN_UART=1 SEL_A=1(TMC220x) DIR=1 VERSION=0x20
Recv: // FACTORY_CONF: 0000000d FCLKTRIM=13
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      000003d8 MSCNT=984
Recv: // MSCURACT:   00f001c4 CUR_A=-60 CUR_B=240
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c00f0000 CS_ACTUAL=15 stealth=1 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  0000007f PWM_SCALE_SUM=127
Recv: // PWM_AUTO:   002000ff PWM_OFS_AUTO=255 PWM_GRAD_AUTO=32

KevinOConnor, 2020-09-22:
There were many important updates in this area since the code version reported in the log.  As you've indicated, upgrading is the best next step.

-Kevin

Roycinger, 2020-09-22:
Hi Kevin,

thanks for replying.
I upgraded via git pull and experienced the same issue again.
Next step I did was setting stealthchop_threshold for all axis drivers to 0 (which disables stealthchop completely) as I think that something was not right there. I had it on 250 (so way above the max print speed) before. 
So far, I didnt have any shut down anymore. I'll report back if this state persists.

Roycinger, 2020-09-25:
Didnt have another shutdown yet, so I'll close this issue for now

closed 2020-09-25 by Roycinger