Klipper3d/klipper #3353: Manual control does not reach maximum

Twilek-de, 2020-09-21:
I have a CR 10 V3 which has a 30x30 bed and 40 z-axis. I have configures the "position_max" accordingly and the printer does print on the whole bed. When I try to manually move the Printhead with the controls on the printer Klipper will only let me move all axis to 200 not 300 or for z 400. I have searched the config options for this but haven´t found anything. What am I missing?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5254355/klippy.log)

FHeilmann, 2020-09-21:
The menu system hardcodes the limits, which is why you're stuck with the default 200 value. You need to override those menu items in your config (copy the relevant blocks of `klippy/extras/display/menu.cfg` into your config and change the values accordingly. The blocks in your `printer.cfg` will then override the default ones in `menu.cfg`

Twilek-de, 2020-09-21:
Worked like a charm, thanks!

closed 2020-09-21 by Twilek-de