Klipper3d/klipper #3355: Skr 1.4 flashing issue

Ducati87, 2020-09-21:
Hello guys, I'm trying to flash the bin file on my skr 1.4 but without success. 
The file bin is on the SD card but when I power it up, marlin boot up. 
Is there a way to flash it? 
Thank you

th33xitus, 2020-09-22:
Did you rename the file to `firmware.bin` ?

Ducati87, 2020-09-22:
> Did you rename the file to `firmware.bin` ?

Yes of course

Ducati87, 2020-09-23:
Any suggest?

Gomer05, 2020-09-26:
Hi @Ducati87 I have a SKR 1.4 Turbo, that just arrived the other day.  It had Marlin 2.0.3 installed from the factory,
Here are the steps that I followed to get Klipper running.  Pretty much followed the docs.
1. built from master branch
**Make sure todo a "make clean", specially if you tried several make menuconfig/make**
I was switching between two different boards and had issues if I didn't run the "make clean"
2. "make menuconfig" and selected:
    Micro-controller Architecture (LPC176x (Smoothieboard))  
    Processor model (lpc1769 (120 MHz))
    [*] Target board uses Smoothieware bootloader
    [*] Use USB for communication (instead of serial)
  Exit and save
3. "make"
4. used the SD card that came with the board.  Put it in my Linux machine and copied ./out/klipper.bin to the root of the card as firmware.bin 
The file was 20Kb in size:  20 -rwxrwxr-x 1 mark mark 20140 Sep 26 17:14 ./out/klipper.bin
5. put the card in the board and powered it up over USB.
6. after afew seconds I had this in the kernel logs showing that the board was now running klipper:
[294691.069044] usb 3-4: new full-speed USB device number 10 using xhci_hcd
[294691.222829] usb 3-4: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[294691.222833] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[294691.222835] usb 3-4: Product: lpc1769
[294691.222838] usb 3-4: Manufacturer: Klipper
[294691.222840] usb 3-4: SerialNumber: 0690000F68944BAF5AA07D5DC22000F5
[294691.224467] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
7. Was able to start klipper and connect to the mcu successfully.

One to check is to make sure the .elf file looks good, you can do:
"file out/klipper.elf" mine looked like: 
out/klipper.elf: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped

closed 2020-09-28 by klipper-gitissuebot

ricochet1k, 2020-11-05:
For those coming after me seeing this issue as a guide to configure the board, note how the "smoothieware bootloader" part is in italic. That's because there are `*` in between the `[]`, which gets read as italic. It should look like this: 

```
Micro-controller Architecture (LPC176x (Smoothieboard))
Processor model (lpc1769 (120 MHz))
[*] Target board uses Smoothieware bootloader
[*] Use USB for communication (instead of serial)
```