Klipper3d/klipper #3356: MCU 'mcu' shutdown: Timer too close (during startup/firmware restart)

dsimbiriatin, 2020-09-21:
Hello,

I haven't updated Klipper for the last few months and with the latest changes I started to get `MCU 'mcu' shutdown: Timer too close` while booting up or calling `FIRMWARE_RESTART`. The issue is not constant, meaning that 1 out of 5 restarts could be successful in the end. I did a small research and found the latest commit where the issue is not present. 

Looks like this commit introduced in some way the regression, at least reverting `mcu.py`  to the state before this commit helps:
[122fd88c6ae6af4f5c485b7b74d3b95d8603df88](https://github.com/KevinOConnor/klipper/commit/122fd88c6ae6af4f5c485b7b74d3b95d8603df88)

[klippy.log](https://github.com/KevinOConnor/klipper/files/5258270/klippy.log)

top-gun, 2020-09-22:
I have encountered the same issue just this night. My klippy.log is zipped as it is beyond the 10MB-limit.

Additional info: 

I tried the print twice, it failed at different positions. It is  huge gcode-file, the object is small but it has lots of very short extrusions. It's one of those moon-lamps that are based on a 3D-model of the surface.

When I looked at the Octoprint-UI, I saw an overheat warning. That hasn't happened in months. My Raspi 3B is usually pretty cool.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/5260278/klippy.zip)

KevinOConnor, 2020-09-22:
There were recent changes in this area (see #3316 / #3331 ).  Upgrade to the latest code (see the FAQ for upgrade instructions) and if the issue occurs again, attach the log from that session.

-Kevin

dsimbiriatin, 2020-09-22:
Hi Kevin,

Thanks for your reply. I've pulled the latest changes according to this guide [How do I upgrade to the latest software?](https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software) and the issue is still there, After the update, I rebooted the RPI and Klipper didn't startup. Here are the logs: [klippy.log](https://github.com/KevinOConnor/klipper/files/5262639/klippy.log)

Not sure if it's important but my setup has:
- Trigorilla board (8-bit), powered via USB from Raspberry Pi
- The power from PSU is controlled through the relay via a pin on the board (which is in HIGH state by default)

KevinOConnor, 2020-09-22:
What type of host machine is the software running on?

-Kevin

dsimbiriatin, 2020-09-23:
It's Raspberry Pi Zero W

JRHeaton, 2020-09-24:
I'm seeing this issue on a Raspberry Pi 4, octoprint with no plugins, fresh OS install with Klipper. Whenever I run a `RESTART` or `SAVE_CONFIG`, any subsequent commands yield a timer too close error and I must run reconnect and run `FIRMWARE_RESTART` to resolve it. I've tried 3 USB cables now, and since all other communications with Klipper (during calibration/setup) worked without disconnection, I'm not convinced it's a connectivity issue (nor a computational strain issue as some other threads have insinuated). Any help would be greatly appreciated, I am new to Klipper.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5273713/klippy.log)

BlackStump, 2020-09-24:
@JRHeaton 
your log error is 
>error: Timeout on wait for 'tmcuart_response' response

try adding - restart_method: command
ie
> [mcu]
>serial: /dev/serial/by-id/usb-Klipper_lpc1769_17D0000D80A037AFD8B5BF5BC62000F5-if00
>restart_method: command

JRHeaton, 2020-09-24:
> 
> 
> @JRHeaton
> your log error is
> 
> > error: Timeout on wait for 'tmcuart_response' response
> 
> try adding - restart_method: command
> ie
> 
> > [mcu]
> > serial: /dev/serial/by-id/usb-Klipper_lpc1769_17D0000D80A037AFD8B5BF5BC62000F5-if00
> > restart_method: command

I tried both `command` and `rpi_usb` just now. The same exact `tmcuart_response` error happens in the log, and same behavior (Timer too close, requires now a `firmware_restart`). I also randomly got a non-reproducible error about connecting to a register on my X driver, only once. Subsequent `restart`s have all had the behavior previously described. Again, I'm totally new to Klipper, so is it possible I could have my drivers configured improperly? What befuddles me with that scenario though is that I am able to, once connected, administer tmc debug commands to read the registers of each driver successfully, so I am doubting that. Just trying to give more data/ideas.

Thanks again for everyone and anyone's help, I am really looking forward to a stable Klipper installation. I've attached my printer.cfg for good measure.

```yaml
# This file contains common pin mappings for the BIGTREETECH SKR V1.4
# board. To use this config, the firmware should be compiled for the
# LPC1768 or LPC1769(Turbo).

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: !P1.29
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
step_distance: .0125
endstop_pin: !P1.28
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_endstop: 0.0
position_max: 300
position_min: -3

[extruder]
step_pin: P2.13
dir_pin: !P0.11
enable_pin: !P2.12
step_distance: .010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: NTC 100K beta 3950
sensor_pin: P0.24
control: pid
pid_kp:16.288
pid_ki: 0.557
pid_kd: 119.104
min_temp: 0
max_temp: 260

[heater_bed]
heater_pin: P2.5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.25
control: pid
pid_Kp: 65.339
pid_Ki: 1.180
pid_Kd: 904.128
min_temp: 0
max_temp: 130

[fan]
pin: P2.3

[bltouch]
sensor_pin: P0.10
control_pin: P2.0
z_offset: 1.600
x_offset: -46
y_offset: -7.5
#pin_up_reports_not_triggered: False

[safe_z_home]
home_xy_position: 120,120 # Change coordinates to the center of your print bed
speed: 50
z_hop: 20                 # Move up 10mm
z_hop_speed: 5

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1769_17D0000D80A037AFD8B5BF5BC62000F5-if00

[printer]
kinematics: cartesian
max_velocity: 400
max_accel: 500
max_z_velocity: 10
max_z_accel: 100

########################################
# TMC2209 configuration
########################################

[tmc2209 stepper_x]
uart_pin: P1.10
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[tmc2209 stepper_y]
uart_pin: P1.9
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[tmc2209 stepper_z]
uart_pin: P1.8
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[tmc2209 extruder]
uart_pin: P1.4
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

########################################
# EXP1 / EXP2 (display) pins
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"
```

JRHeaton, 2020-09-24:
From the above, it seems like whatever mechanism Klipper uses for restarting and reconnecting is either somewhat broken at the moment, or I don't understand what's really happening. To summarize:
- Klipper initially connects correctly, but when I inevitably send a `restart` it produces the tmc uart log error and the "Timer too close" error in OctoPrint.
- As a result, I have to use `firmware_restart` to recover.
- `firwmare_restart` stops working occasionally, the serial port in `/dev/serial/by-id/` disappears, and I must plug the USB cable into a different port for it to show again. I have no idea why this would be... I even tried restarting the Klipper service. Has to be a different hardware port.

Again, any further guidance on how to debug, test, or otherwise help in fixing this is appreciated. In this state, I don't think I can use Klipper reliably.

KevinOConnor, 2020-09-25:
@dsimbiriatin - if you find the two lines in klippy/mcu.py that have:
```
self._last_clock = self._mcu.print_time_to_clock(printtime + 0.100)
```
And change them to:
```
self._last_clock = self._mcu.print_time_to_clock(printtime + 0.200)
```
And then run:
```
sudo service klipper restart
```
Does the problem go away?

-Kevin

david-sawatzke, 2020-09-26:
I'm having the same/very similar issue on a pi model b (the same soc), with a *very* similar looking log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/5287405/klippy.log). The only reason I'm not 100% percent sure it's the same issue is 'cause it doesn't seem intermittent in my case, it always fails.

I've tried to adjust the `self._last_clock` line, but it didn't help, the problem persisted.

dsimbiriatin, 2020-09-27:
@KevinOConnor, I can confirm that it helps, thanks!

KevinOConnor, 2020-09-28:
@david-sawatzke - if you deploy the same solution described at https://github.com/KevinOConnor/klipper/issues/3356#issuecomment-698687562 does the problem go away?

-Kevin

dsimbiriatin, 2020-09-28:
@david-sawatzke, did you update both occurrences (line 218 and line 243)?

david-sawatzke, 2020-09-28:
@dsimbiriatin Should read more carefully, only updated one before, thanks!

@KevinOConnor It now works!

KevinOConnor, 2020-10-11:
Okay, I committed that change to the master branch (commit 27249dc7).

-Kevin

dsimbiriatin, 2020-10-11:
@KevinOConnor Thank you for helping with the issue!

closed 2020-10-11 by dsimbiriatin