Klipper3d/klipper #3357: Single heater, multiple thermistor. Enhancement request.

weaslus, 2020-09-21:
I'd like to request added functionality to support 2 thermistors or more for 1 heater. Here's the jist of it.

To allow for more accurate temperature readings at the surface of a build plate, I'd like to use 1 thermistor installed at the edge of the build plate to tune PID and inform the heater.
Secondarily, as my heater has a built in thermistor, it would be good to be able to use this as a watchdog, to make sure if the temp ever exceeds a user defined maximum, it can interrupt the PID cycle until the temperature has dropped back to with the defined safety margin.

This double thermistor functionality, could double as a backup safety measure. Should the heater decide to jump off the build plate.

I know it's currently possible to define 1 thermistor just to max temp, but that would lead to prints failing, where it could just be a case of waiting for the temp to come down before continuing to heat.

Although I believe it's possible to code some of this functionality into a macro, I'm concerned calling for the heater to cut out should the watchdog thermistor read over temp, would lead to erratic behaviors tripping some thermal protection. And seems to be a deeper coding task than I'm capable of.

Please let me down nicely if this is best done by a macro. It is however a function I'd like to run continuously should it be chosen, and i'm not sure if it can as a macro.

Thanks.

weaslus, 2020-09-22:
It's been communicated to me, that running 2 thermistors for logic control is not something that's done in this method by anyone yet, in any field. So effectively I've just requested you pioneer new logic and handling of PID. I am therefore closing the issue.

arne182, 2021-08-21:
I recently invested in the Slice Mosquito Magnum + Hotend. Which has this exact requirement. That and it has two heater cartridges. So what I would think needs to be done is link the two heaters. And average the thermistors unless they read more than a certain value different from one another and then could throw an error. I would gladly help with the coding if you can point me in the right direction.

weaslus, 2021-08-22:
As above. Re-opened for slice magnum plus with dual heater cartridges and dual thermistors. This is now beyond my wheelhouse and i will leave this to @arne182 to lead the path.

closed 2021-09-20 by github-actions[bot]