Klipper3d/klipper #3359: How to move the extruder without thermistor. MCU 'mcu' shutdown: ADC out of range

acastles91, 2020-09-22:
Hello. I am using klipper with an SKR 1.4 for a custom build. It's a pen plotter with an airbrush, in which the extruder controls the needle of the airbrush. I don't need a thermistor for my build; instead, I want to use a regular endstop to set the limits of the movement of the needle. I am getting the ADC out of range error, even though I've set the min_extrude_temp value to 0. What is the best way  to ignore all the warnings related to temperature? Is there a way to incorporate an endstop to the extruder?

Thank you for your help.

[
[klippy.log](https://github.com/KevinOConnor/klipper/files/5262430/klippy.log)
](url)

KevinOConnor, 2020-09-23:
You can set `min_temp=-273.15`, `max_temp=9999999999`, and `min_extrude_temp=-273.15`.

Separately, it may also be possible to use a manual_stepper config section instead of a fake extruder to control your airbrush.

-Kevin

acastles91, 2020-09-23:
Thank you very much for your reply @KevinOConnor , Is there any example for the manual_stepper config that I should look at? Does that mean that I would have to assign pins manually? I would like to try both alternatives. Thanks!

jakep82, 2020-09-24:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L812

acastles91, 2020-10-01:
Thanks again for your help. I set the temperature to the indicated vales and it works, but I still need to home the stepper, since the needle of the airbrush that I'm using cannot move indefinitely and I want to work within a well defined range. Is there a way to combine an endstop with an extruder? I tried defining the pins of the extruder as a manual stepper but as I understand, that means that I have to work with the MANUAL_STEPPER commands, which don't take place inside the same gcode line as the other coordinates do.

KevinOConnor, 2020-10-03:
There is no ability to "home an extruder".  You can home a manual_stepper.  You are correct that manual_stepper commands are issued separately from G1 commands.

FWIW, I don't know enough abut your airbrush setup (or any airbrush for that matter), so there isn't much assistance I can offer.

-Kevin

acastles91, 2020-10-03:
I'm also new to airbrushes, but long story short, the main difference between that and an extruder is that the needle that regulates the amount of paint that comes out of the nozzle can only move within a range, in my case from 0 to 60. So my need to home the extruder comes from that. I'm thinking about developing a routine of putting the needle manually in the 0 position when the machine is off, and then do a G92 E0 right away, and configuring my gcode script to always return to E0 at the end of each "print". Do you think it would make sense? I know it might be a very specific use case but, with that information in mind, if you could give me your opinion it would be amazing. Thanks for your help.

KevinOConnor, 2020-10-11:
FWIW, a new host python module (eg, klippy/kinematics/airbrush.py) might make sense.  If so, an interested developer would need to take that on and submit it when ready.

-Kevin

closed 2020-11-09 by github-actions[bot]