Klipper3d/klipper #3360: BLTouch Take Multiple Samples to set z-stop

Rneil-ca, 2020-09-22:
It would be nice if we where able to set the number of samples that the BLTouch uses to set the z-stop, I have not been able to find a way to set this.  I have a saved bed_mesh which I load when I print, it is a 5x5 mesh created with 4 samples at each point.  The limitation to this is that the z-stop only takes one rough and 1 fine sample when setting the z-stop, you should be able to take multiple samples and have it use the average as the z-stop.   Please let me know if there is a way to do this, as I have not been able to find one.

Thanks,
Ryan
[klippy.log](https://github.com/KevinOConnor/klipper/files/5262684/klippy.log)

Tahutipai, 2020-09-22:
Have you checked the config options [here]?(https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg)

Start at section "[probe]", currently on line 620.  You can set the number of samples, whether to use Average or Median, and other options.

Does that answer your query?

Sineos, 2020-09-22:
Actually I fully agree with this request:
- Multiple samples work fine for **creating** the mesh
- For the Z-**homing** process basically only one true sample (quick approach and one true sample) is taken

Given the inherent inaccuracies of the Probes this might really make a difference.
Actually the same is true for other Probe related functions like `PROBE_CALIBRATE` or `SCREWS_TILT_CALCULATE`

Rneil-ca, 2020-09-22:
For PROBE_CALIBRATE Klipper does use the samples: # from the [bltouch] in the config, but it does not use it for z-homing.

ProfEngr, 2020-09-23:
Am I missing something? Honestly, I'm not being smart-a$$.

My v2 BLTouch just returned this (15 samples):

Recv: // probe accuracy results: maximum 1.052500, minimum 1.047500, range 0.005000, average 1.051500, median 1.052500, standard deviation 0.001528

A full Z step on my Ender 3 is 0.0025. The range may be 2 full steps, but that's an order of magnitude smaller than even a 0.08 layer height.

Rneil-ca, 2020-09-23:
Hi ProfEngr,

My probe is not as accurate as yours:
probe accuracy results: maximum 0.056500, minimum 0.031500, range 0.025000, average 0.043250, median 0.042750, standard deviation 0.006987

The point is, we are doing the whole print just based on one sample, if that sample is off, then the first lay could be either compressed or too high and not stick well.  The reason they have the option to do multiple probes at each point when doing a bed mesh is to correct for these deviations in the probe. 

Do you do multiple probes at each point when you do your bed mesh? If so why?

jakep82, 2020-09-23:
Averaging the results of an inaccurate probe isn't necessarily going to make the results more accurate.  The [Klipper docs](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#repeatability-check) recommend a probe that has a range of no more than 25 microns, and it appears your probe is just within that limit.  Slowing down the second_homing_speed is probably a better solution to improve your probe accuracy.

ProfEngr, 2020-09-23:
@Rneil-ca 
I do multiple probes when meshing, but only because the option was there and I said what the heck. I also only run a new mesh maybe every month at the most frequent. A 9x9 doesn't take too long when it's that infrequent.

Rneil-ca, 2020-09-24:
@ProfEngr 
I do the same thing for my mesh, but again you're first layer is dependent on the 1 probe result for you z stop when you print, if this one happens to be on the high end of the spec, it could mess up your first layer.   

@jakep82
I agree that averaging doesn't fix an inaccurate probe but it does help if you do happen to get one result that is at the far end of the spec. If that one result was the one you used to set your z stop, you could have issues.

Just because some people don't see a need I think that many people would agree that this would be helpful.  They allow multi sampling while doing the bed mesh, for this exact same reason, so I don't think this was an unreasonable ask.  Originally the bed mesh was done at the beginning of every print so this would not have been a issue, it only becomes a problem when using a saved mesh.

KevinOConnor, 2020-09-25:
There currently is no way to "home multiple times" for improved endstop accuracy.  Similar items have been raised in the past (eg, #2167 ).

Ultimately, an interested developer would need to take this on and submit the code.  (FWIW, it is currently a little tricky to rework the homing code to support some of the functionality of the probe code.)

-Kevin

schnello, 2020-10-06:
> (FWIW, it is currently a little tricky to rework the homing code to support some of the functionality of the probe code.)

As a work arround a dev could implement that the z result from "PROBE" could be processed as variable in an macro like the "query_probe" and "printer.probe.last_query" attribute. I know this is not good solution but better as nothing.

closed 2020-11-05 by github-actions[bot]