Klipper3d/klipper #3363: Klipper v0.9.0 release

KevinOConnor, 2020-09-22:
This issue is notice that I am planning on making the next release of Klipper (v0.9.0). My rough plan is to make the release in mid-October, with a feature freeze starting at the end of September.

If anyone is aware of any critical bugs not already reported, please open a new github issue with a report of the issue. With the start of the feature freeze, the Klipper master branch will focus on bug fixes and documentation updates.

-Kevin

Mechazawa, 2020-10-06:
Is there a list of changes between 0.8 and 0.9 or should I just look at the git log?

KevinOConnor, 2020-10-13:
I will update the Release notes with high-level changes as part of the release.  https://www.klipper3d.org/Releases.html

My draft notes are:

Klipper 0.9.0
=============

Available on 202010??. Major changes in this release:
* Support for "Input Shaping" - a mechanism to counteract printer
  resonance. It can reduce or eliminate "ringing" in prints.
* New "Smooth Pressure Advance" system. This implements "Pressure
  Advance" without introducing instantaneous velocity changes. It is
  also now possible to tune pressure advance using a "Tuning Tower"
  method.
* New "webhooks" API server. This provides a programmable JSON
  interface to Klipper.
* The LCD display and menu are now configurable using the Jinja2
  template language.
* The TMC2208 stepper motor drivers can now be used in "standalone"
  mode with Klipper.
* Improved BL-Touch v3 support.
* Improved USB identification. Klipper now has its own USB
  identification code and micro-controllers can now report their
  unique serial numbers during USB identification.
* New kinematic support for "Rotary Delta" and "CoreXZ" printers.
* Micro-controller improvements: support for stm32f070, support for
  stm32f207, support for GPIO pins on "Linux MCU", stm32 "HID
  bootloader" support, Chitu bootloader support, MKS Robin bootloader
  support.
* Improved handling of Python "garbage collection" events.
* Many additional modules added: adc_scaled, adxl345, bme280,
  display_status, extruder_stepper, fan_generic,
  hall_filament_width_sensor, htu21d, homing_heaters, input_shaper,
  lm75, print_stats, query_adc, graph_accelerometer, graph_extruder,
  graph_motion, graph_shaper, graph_temp_sensor, whconsole
* Several bug fixes and code cleanups.

KevinOConnor, 2020-10-20:
The 0.9.0 version of Klipper has now been released! For more information on the release, please see:
https://www.klipper3d.org/Releases.html

There is a Patreon page for those who want to donate to Klipper:
https://www.patreon.com/koconnor

New in this release:
* Support for "Input Shaping" - a mechanism to counteract printer
  resonance. It can reduce or eliminate "ringing" in prints.
* New "Smooth Pressure Advance" system. This implements "Pressure
  Advance" without introducing instantaneous velocity changes. It is
  also now possible to tune pressure advance using a "Tuning Tower"
  method.
* New "webhooks" API server. This provides a programmable JSON
  interface to Klipper.
* The LCD display and menu are now configurable using the Jinja2
  template language.
* The TMC2208 stepper motor drivers can now be used in "standalone"
  mode with Klipper.
* Improved BL-Touch v3 support.
* Improved USB identification. Klipper now has its own USB
  identification code and micro-controllers can now report their
  unique serial numbers during USB identification.
* New kinematic support for "Rotary Delta" and "CoreXZ" printers.
* Micro-controller improvements: support for stm32f070, support for
  stm32f207, support for GPIO pins on "Linux MCU", stm32 "HID
  bootloader" support, Chitu bootloader support, MKS Robin bootloader
  support.
* Improved handling of Python "garbage collection" events.
* Many additional modules added: adc_scaled, adxl345, bme280,
  display_status, extruder_stepper, fan_generic,
  hall_filament_width_sensor, htu21d, homing_heaters, input_shaper,
  lm75, print_stats, resonance_tester, shaper_calibrate, query_adc,
  graph_accelerometer, graph_extruder, graph_motion, graph_shaper,
  graph_temp_sensor, whconsole
* Several bug fixes and code cleanups.

     
For further information on Klipper, please see:
https://github.com/KevinOConnor/klipper

luco85, 2020-10-21:
Thanks kevin!

KevinOConnor, 2020-10-28:
Due to a couple of bugs that were found after the v0.9.0 release, I have made a new v0.9.1 release.  This release contains only bug fixes (notably, the pressure advance "numerical stability" fix, and the bltouch "stow_on_each_sample=false" fix).

-Kevin

closed 2020-11-13 by KevinOConnor