Klipper3d/klipper #3364: Negative un-retraction

sedlacekdavid, 2020-09-23:
Hi, is it possible to configure a small negative un-retraction move during tool move and retractions?

I've seen it on the internet and I was kind of hoping to solve my small stringing issues. I'm using a bit longer nozzle (a Volcano hotend). I haven't found this it in cfg files nor here.

Thank you, David

jakep82, 2020-09-23:
I'm not sure what you mean by negative unretraction, but this is something I would expect to configure in a slicer, not in firmware settings.

Hywelmartin, 2020-09-23:
negative (-) unretraction(opposite of  - =  + ) = retract 
https://github.com/KevinOConnor/klipper/blob/08eec0da86b73db7190fb26fc17199d6eeb30967/config/example-extras.cfg#L488
Or use the slicer

sedlacekdavid, 2020-09-24:
I'd like to set negative unretract_extra_length, but it's not possible since there is a negative value check preventing setting it to a negative number.

Hywelmartin, 2020-09-24:
I'm not a coder but this seems to be the problem
https://github.com/KevinOConnor/klipper/blob/08eec0da86b73db7190fb26fc17199d6eeb30967/klippy/extras/firmware_retraction.py#L38

sedlacekdavid, 2020-09-24:
Yep that's clear that there would be a check like that. But it's a question how many submodules relies on a fact that the number is non-negative.... So it can be more complicated than changing that assert. 

I'm not saying that this would help solving my PETG stringing with a long Volcano hotend, but there are many suggestions on the internet that it does help setting a small negative value like -0.2 mm or so.

Hywelmartin, 2020-09-24:
Negative unretract_extra_length does not help with stringing.. high acceleration, speed, PA and a dry material helps..

DroneMang, 2020-10-06:
Try this, I asked the same question a while ago if I understand you correctly

Hi @DroneMang
--
in your rapsberry pi go to the folder Klipper > Klippy > extras
In firmware_retraction.py try changing the following:
Line 13, make the minval a negative number that meets your needs
Line 41 make the same change
 
save the file and restart the pi. Test it out and see if it works effectively without causing errors or other print problems.
 
David

closed 2020-11-05 by github-actions[bot]