Klipper3d/klipper #3366: Neopixel from 'linux process' gpio seems broken

buzztiaan, 2020-09-25:
Some findings about the state of neopixel for the 'linux process' mcu:

* initial_XX values seem ignored, the printer now -always- has all leds on 100% regardless of config
* chain_count is maximum 18? i have 20 leds on my printer and expecting to add more :D
* **SET_LED LED=toplights RED=1.0 GREEN=0.0 BLUE=0.0** commands do nothing to change the color of the leds

printer.cfg snippet : 
```
[mcu pihost]
serial: /tmp/klipper_host_mcu

[neopixel toplights]
pin: pihost:gpio18
chain_count: 18
initial_RED: 0.0
initial_GREEN: 1.0
initial_BLUE: 1.0
```

buzztiaan, 2020-09-25:
Added the log here, although i really don't see anything neopixel related in it : 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5283137/klippy.log)

KevinOConnor, 2020-09-28:
Unfortunately, the neopixel functionality doesn't currently work on the "linux mcu" implementation of Klipper.

As we discussed on IRC, the neopixel requires very strict timing for its bit-banging interface.  Specifically, a zero bit must be transmitted with a gpio pulse of no more than ~500ns.  The "linux mcu" GPIO code uses the linux kernel gpio interface which isn't nearly fast enough to reach that rate.  (In my tests, two back-to-back "toggle" commands took ~11us - way more than the required 500ns.)

Unfortunately, I don't see a simple solution.  It might be possible to obtain the necessary timing if the mcu code did direct register writes for gpio updates.  However, that would require a lot of work.

-Kevin

buzztiaan, 2020-09-28:
Maybe some time in the future then :) Issue closed

closed 2020-09-28 by buzztiaan

NameOfTheDragon, 2021-01-30:
The OctoPrint LED Status plugin somehow manages it with a DMA channel. This is a bit of a black eye for Klipper. Please re-open this issue @KevinOConnor

buzztiaan, 2021-03-18:
it seems they use SPI? or i didnt find the right plugin

buzztiaan, 2021-11-11:
> It would be excellent if ..
You can edit that file here ; https://github.com/Klipper3d/klipper/blob/master/docs/Config_Reference.md

schnello, 2022-01-06:
@Nicholas-Barnes-Ohio 
I had the same problem.
https://github.com/Klipper3d/klipper/pull/5113

buzztiaan, 2022-01-06:
offtopic a bit, eventually i'll just add a arduino something on USB to the pi , and run it as second mcu, just for the LEDs (and maybe some other stuff)