Klipper3d/klipper #3367: Custom menu entries disappeared

Roycinger, 2020-09-25:
Hi,

After my latest git pull (around 4 days ago), I edited my menu.cfg to add some custom points.
They showed up and everything was working.

Now I just noticed that my custom menu point "Load / Unload" is no more present.
My menu.cfg:
https://pastebin.com/66Ku8NCP

Any reason why those menu point is gone?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5283437/klippy.log)

ld3300, 2020-09-25:
I'm behind myself, and will be spending the next couple of days catching up, but in [klippy/extras/display/menu.cfg](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/display/menu.cfg) it looks like the format has changed a little.
try changing:

`!toolhead.is_printing`

to:

`{not printer.idle_timeout.state == "Printing"}`

-David

Roycinger, 2020-09-25:
Hi David,

that's highly prossible indeed.
I expected Klipper to throw an error when using an old syntax (like it did when the menu.cfg had the "old" format).
I'll try after the running print and report back.
-Paul

ld3300, 2020-09-25:
Hi Paul,

My guess is that Klipper sees the logic as a valid command, but it just always returns false.  So it probably can't tell there's a problem.  Please let me know if it works, I'm no pro, and appreciate opportunities to learn.

-David

Roycinger, 2020-09-25:
I'll report back, will most likely be tomorrow after that 10hr print :)

Roycinger, 2020-09-26:
The syntax that David mentioned worked indeed.
Thanks for pointing that out!

-Paul

closed 2020-09-26 by Roycinger