Klipper3d/klipper #3369: Klipper shuts down when heating my bed to 100C

ThatOneGuy030, 2020-09-26:
This is the first time i'm pushing my bed to heat this high and it reaches PLA temps just fine. Now that I tried 100C for ABS klipper keeps shutting down at around 85C. When heating for a print it also says that the bed wasn't heating at an expected rate. Sorry if the format is bad made an account just for this.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5286222/klippy.log)

Gomer05, 2020-09-26:
When you say that the bed has been modified, did you do a PID tune after the modification?  Sometimes it helps to retune to the higher if you are shooting for temps higher than the current PID tune is at.

ThatOneGuy030, 2020-09-26:
> When you say that the bed has been modified, did you do a PID tune after the modification? Sometimes it helps to retune to the higher if you are shooting for temps higher than the current PID tune is at.

I have not modified the bed in any way (unless replacing the glass counts). I have changed the post to better convey this.

Hywelmartin, 2020-09-27:
`Heater heater_bed no longer approaching target 100.000
Stats 793.0: gcodein=640 mcu: mcu_awake=0.009 mcu_task_avg=0.000095 mcu_task_stddev=0.000071 bytes_write=10730 bytes_read=35270 bytes_retransmit=9 bytes_invalid=0 send_seq=956 receive_seq=956 retransmit_seq=17 srtt=0.010 rttvar=0.005 rto=0.029 ready_bytes=0 stalled_bytes=0 freq=16000354 heater_bed: target=100 temp=84.6 pwm=1.000 print_time=0.000 buffer_time=0.000 print_stall=0 extruder: target=0 temp=30.6 pwm=0.000
Heater heater_bed not heating at expected rate`

so a more powerful bed heater or a dulled down safety check..https://github.com/KevinOConnor/klipper/blob/55e20f2c198ae45485d714b4588fcf6768bc6165/config/example-extras.cfg#L841

closed 2020-10-27 by github-actions[bot]