Klipper3d/klipper #3373: SPI DAC DAC084S085 implementation

Blaster1920, 2020-09-26:
LOG FILE: [klippy.log](https://github.com/KevinOConnor/klipper/files/5286653/klippy.log)

Hi, I'm trying to communicate to my SPI DAC that controls the motor drivers's VRef on my Alligator rev.2 board.

The DAC is a DAC084S085 and here's the implementation for RepRapFirmware (RRF):
[DAC084S085 RRF.zip](https://github.com/KevinOConnor/klipper/files/5286657/DAC084S085.RRF.zip)

So far with the help of @MarcoAntonini we managed to generate the values that needs to be used to set the DAC channel and the DAC value (then translated into vref).
[dac_alligator py.txt](https://github.com/KevinOConnor/klipper/files/5286664/dac_alligator.py.txt)

Unfortunately as you can see from the .log it looks like the custom subsection shouldn't be in mcu(?)
What's would be the correct implementation for this?

Thanks in advance and congratulation for the immense work you've done so far.

KevinOConnor, 2020-09-27:
The `custom` option was removed some time back (it was too cryptic to use or maintain).

I recommend you take a look at klippy/extras/ad5206.py - you'll want to copy that to klippy/extras/dac084S085.py and then make the changes necessary to set the 4 channels on your particular device.

-Kevin

Blaster1920, 2020-09-29:
> The `custom` option was removed some time back (it was too cryptic to use or maintain).
> 
> I recommend you take a look at klippy/extras/ad5206.py - you'll want to copy that to klippy/extras/dac084S085.py and then make the changes necessary to set the 4 channels on your particular device.
> 
> -Kevin

Hi,
I tried to do as suggested but the SPI bus is not being set. I used a signal spoofer to check the busses but they're empty.
This is the script I've written:
[dac084S085.txt](https://github.com/KevinOConnor/klipper/files/5302091/dac084S085.txt)

And this is how it's called from the config file:
`[dac084S085 my_digipot]
enable_pin: PB14
vrefX: 125
vrefY: 125
vrefZ: 125
vrefE: 125
`
Any suggestion? With RRF I can confirm the board is functional.

KevinOConnor, 2020-10-02:
Not sure why that would be.  I'd say double check your pins.  You can also use the console.py tool to manually issue commands for diagnostic purposes (though it's a bit of work to configure the spi connection before you can send commands on it).  https://www.klipper3d.org/Debugging.html#manually-sending-commands-to-the-micro-controller

-Kevin

Blaster1920, 2020-10-07:
> Not sure why that would be. I'd say double check your pins. You can also use the console.py tool to manually issue commands for diagnostic purposes (though it's a bit of work to configure the spi [...]

Ok, managed to fix the code, it is now working as expected and tested on the board: 
[dac084S085.txt](https://github.com/KevinOConnor/klipper/files/5343930/dac084S085.txt)

usage:
`
[dac084S085 my_digipot]
enable_pin: PB14
spi_bus : spi0
motor1 : 10
motor2 : 100
motor3 : 150
motor4 : 255
`

Is there any chance you could commit it?

KevinOConnor, 2020-10-07:
Great.  It looks like useful functionality to me.  It would need to be submitted as a "github pull request" for me to commit - see: https://github.com/KevinOConnor/klipper/blob/master/docs/CONTRIBUTING.md .  FYI, I'd recommend some code changes as part of that review (eg, add copyright, time.sleep() isn't valid, add scale config parameter, add docs to config/example-extras.cfg).

Cheers,
-Kevin

PR #5134 (2022-02-22): SPI DAC DAC084S085 implementation
SPI DAC DAC084S085 integration to support Alligator Board

Fixes: #3373

@KevinOConnor, I totally agree with the licensing terms described at https://github.com/KevinOConnor/klipper/blob/master/docs/developer-certificate-of-origin

This is the integration of the SPI DAC DAC084S085 that controls the motor drivers's VRef on my Alligator rev.2 board made by Jacopo Franco and Marco Antonini #3373.
New version tested of PR #3840.
I am attaching the klippy log (for now it is being tested on a Raspberry Pi B so with very few resources).
[klippy.log.txt](https://githu [... cut]
closed 2020-11-05 by github-actions[bot]