Klipper3d/klipper #3374: Fans nonfunctional

Zwaubel, 2020-09-27:
Since 55e20f2c198ae45485d714b4588fcf6768bc6165the fans are no longer working. Neither an additional fan nor the standard extruder fan spins up automatically or via M106 command.
I might be missing something and if so I'm sorry for spamming the issues board.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5288505/klippy.log)

ld3300, 2020-09-27:
Hi @Zwaubel,

I'm not able to reproduce the problem on my end.  Will you try a couple of things for me?  Does the hotend fan kick on if you set the hot end to 100 degrees?  Will you comment out the Hardware PWM line and see if that makes a difference?

thanks,

-David

Zwaubel, 2020-09-27:
Hi David,

Thank you for looking into this. 
I was fiddling around with the temperature setting already. From within the normal operating range (20 C (cold) - 215 C (printing temp)) the temperature does not seem to affect the fans behavior.
I will try using SW PWM later today or tomorrow depending on my time schedule and will report back asap. 

Thanks anyway, 
Vincent.

KevinOConnor, 2020-09-27:
There was an error in a recent merge.  I think it should be fixed now (commit 4d0d2197).

-Kevin

Zwaubel, 2020-09-28:
Hi Kevin,

I'll check that today, when I'm back at the printer and will report back. 
Thanks for those quick adjustments. 

Vincent

Zwaubel, 2020-09-28:
Update: both fans are working again, but strangely my [heater_fan hotendFan] won't accept hardware PWM anymore. Will crosscheck the corresponding arduino pins HW PWM support tomorrow. But I'm pretty sure I was using it in HW PWM mode before...

Zwaubel, 2020-09-29:
I'm going to close this as the printer is working fine despite the hardware pwm issue on

closed 2020-09-29 by Zwaubel