Klipper3d/klipper #3375: [FR] Cancel object functionality

MarcPot, 2020-09-27:
Recently I went over to Mainsail from Octoprint and it works great! Big improvement in combination with Klipper. The only feature I’m really missing now is the “Cancel Object” functionality. So this issue is a feature request to ask for this feature.

I already talked to the Mainsail and Moonrakers devs and they told me that this is a Klipper thing, not a Mainsail/Moonraker thing.

Arksine, 2020-09-27:
FWIW, this is going to require support from both Mainsail and Klipper.  Mainsail needs a gcode viewer for this kind of functionality to be useful.  I have plans to add support for exclude region in the future, but I don't have a definitive timeline.

MarcPot, 2020-09-27:
Well sure for the exclude region kind of functionality, but I think mist people should be fine with a simple list of objects and then the option to click cancel next to the object.

I just use superslicer so I don’t know about other slicers, but superslicer has the function to define objects.

ProfEngr, 2020-09-29:
Both Region and Object exclusion are done with Octoprint plugins and don't care about your firmware. I've used both on Marlin and Klipper. Mainsail should be the one responsible for the function if it's even remotely similar to Octoprint in that it just feeds gcode to the firmware. I'm probably talking out of turn as I have only hear Mainsail in passing. (off to do some research now)

MarcPot, 2020-09-29:
> Both Region and Object exclusion are done with Octoprint plugins and don't care about your firmware. I've used both on Marlin and Klipper. Mainsail should be the one responsible for the function if [...]

Mainsail and Moonraker are just a webinterface and a webinterface api, it doesn't actually do anything with the Gcode, that's handled by Klipper and the reason it has to be integrated into Klipper.

Arksine, 2020-09-29:
> Both Region and Object exclusion are done with Octoprint plugins and don't care about your firmware. I've used both on Marlin and Klipper. Mainsail should be the one responsible for the function if [...]

Moonraker (the backend) does not directly parse gcode files like Octoprint.  Instead it exposes APIs with which clients (such as Mainsail) can access Klipper's `virtual_sdcard` functionality.   Octoprint's gcode parser allows plugins like Exclude Region and Cancel Object to intercept gcode commands and exclude them if they fall within an excluded area.  The downside to this that there is additional overhead from the gcode processing.   Moonraker has very little overhead, thus it can conceivably run together with Klipper on a Pi Zero.

Fortunately Klipper's own gcode parser exposes APIs with which we can transform gcode commands.  Writing an extras module to exclude an area/object wont be that difficult outside of dealing with retractions.  However I still believe that without a UI using these kinds of modules would be an exercise in frustration.   Consider a plate 10 copies of an object, where one object fails.  A simple list of objects is probably not going to be good enough for a user to know which object they need to cancel. 

I would concede that a full gcode analyzer probably isn't necessary for a usable "cancel object" UI.  The object metadata includes bounding coordinates, so a simple UI with selectable "boxes" would probably be good enough for cancel object.

ProfEngr, 2020-09-29:
Yeah, I should have looked up the other gits before hitting the keyboard here. *facepalm*

BlackStump, 2020-09-29:
I personally like the fact that Klipper-Moonraker and Mainsail is that efficient to can be run on a Pi Zero.

What I would not want to see is Klipper-Moonraker-Mainsail loose that efficiency to become Octoslow's cousin.
People who want all the bells and whistles multiple useless plugins have a option and it is Octoprint. 

I would hope if any addition features are implemented in Moonraker-Mainsail that they include the option to disable them if not required and-or they have a overhead burden.

I never quite got the gcode visualizer, I load a model into the slicer, I slice it and name it appropriately so I know exactly what it is by the name, why do I need to see it again in mainsail. I do not use superslicer but there thumbnails seem like a good option for people that want to visualizer their gcode otherwise drop it back into the slicer that is it's job.  

I am a function over form person. efficiency first. Of course my post can be totally disregarded as it is purely my personal opinion.

Hywelmartin, 2020-10-01:
If objects are defined in the slicer( assume every row??)..  then its just a wait for the object that should be cancelled is getting printed.. no graphical UI is needed, more than a button that cancels the ongoing object

randellhodges, 2020-10-03:
I think you'd need tie into the virtual sdcard code to implement this. How would you expect this to work? I might take a stab at it in my free time.

When the file is loaded with M23, should it, at that time, scan the file and find/output a list of found objects, or, would you want another command to be executed to do that?

If you went the 2nd way, the flow would be something like (command names to be determined)

M23 file.gcode
LIST_OBJECTS
-> object_1
-> object_2
M24
CANCEL_OBJECT NAME=object_1

Or would you want to make cancel use an index?

I'm thinking the LIST_OBJECTS (or the M23 when loaded) would scan the file and build a dictionary/position of objects in the file. Klipper would then keep track of which object is currently printing. When you request to cancel it would mark that object as cancelled and then when the layer is complete, the next time it "enters the object" it would skip lines until it gets to the next, non-cancelled object.

Would it be assumed you could ignore every command "between" the objects?

I haven't dug thru the code yet, but, could an extras module access the gcode that was loaded by M23, or would the virtual_sd module need to expose/provide access to the new extras module?

Arksine, 2020-10-04:
@BlackStump This would be completely optional.  That said, I think its implementation would be reasonably efficient and wouldn't add much overhead. 

>If objects are defined in the slicer( assume every row??).. then its just a wait for the object that should be cancelled is getting printed.. no graphical UI is needed, more than a button that cancels the ongoing object

This could be possible, but lookahead might cause some issues with this approach.   Cancel an object too late and it will cancel the wrong one.

>I think you'd need tie into the virtual sdcard code to implement this. How would you expect this to work? I might take a stab at it in my free time.

I don't plan to make any changes to the virtual sdcard.  While Moonraker does not process gcode files in realtime, it does have a metadata parser that analyzes each file when its uploaded.  It won't be a problem to extend it to extract Object headers.

For the real time processing, Klipper's `gcode.py` checks for comments and ignores them.  We just need to add a hook there so the "cancel_object" module can determine if the comment contains an object identifier.   In addition to this, we can register a move transform that will "skip" moves if a canceled object/region is detected.  The tricky part is handling retractions and detractions.  Additional logic will be necessary to ensure that neither is done consecutively.

MarcPot, 2020-10-04:
What Arksine said is completely true, just a list of components would probably not be enough, but a list of components with a bounding box as position indicator would probably be enough in 99% of the cases.

Insert this functionality with a [cancel_object] in your printer.cfg would just add this function when you ask for it so it deals with the slowness, but I fail to grasp why this function would be cpu intensive since you just have to skip a few lines?

When Superslicer makes a gcode file every part will be done separately every layer (after layer 1).
When the last part of the line is laid Superslicer inserts a 

> ; printing object Shape-Box id:0 copy 0

This will be followed with the retract and z-hop or whatever. So you just have to skip everything between the first (; printing object) and the next (; printing object). The way I read this gcode file, you dont have to do anything but skip those lines.
[Shape-Box.zip](https://github.com/KevinOConnor/klipper/files/5323214/Shape-Box.zip) Here a simple example with 5 copied boxes.

> ; stop printing object Shape-Box id:0 copy 4
> ;BEFORE_LAYER_CHANGE
> ;0.4
> 
> 
> G1 Z0.400 F18000.000
> ;AFTER_LAYER_CHANGE
> ;0.4
> ; printing object Shape-Box id:0 copy 0
> 

It even puts the non object related gcode outside of the (; printing object)

Hywelmartin, 2020-10-06:
> This could be possible, but lookahead might cause some issues with this approach. Cancel an object too late and it will cancel the wrong one.

It depend on if you cancel ongoing(processed in the look ahead que) or cancel the stated ongoing print.. ( so the status need to be stated.. now printing part 4..)  sort of ...

DroneMang, 2020-10-06:
Curious, I used to use cancel objects with Marlin and Octoprint, but I havent got it working with Klipper yet.  What do you have to configure with Klipper and or  your slicer (S3D)?

closed 2020-11-05 by github-actions[bot]

robthide37, 2020-12-02:
Wish theis never lost steam as it really is a great idea

Arksine, 2020-12-02:
I have plans to work on this, but I have no timeline as I'm not sure when I will have time to get to it.

DanielJoyce, 2020-12-14:
Klipper firmware just controls the printer, let's keep it simple and not do gcode processing to ignore objects in the gcode. Otherwise you keep adding feature and you're back at Octopi.

To me the place for this should be a middle ware, so Mainsail, or Fluidd, could offer the ability to parse the gcode, remove the objects, and then send it to klipper. Let's keep klipper at 'parse gcode, control printer'.

MarcPot, 2020-12-14:
@DanielJoyce 

> 
> 
> > Both Region and Object exclusion are done with Octoprint plugins and don't care about your firmware. I've used both on Marlin and Klipper. Mainsail should be the one responsible for the function if it's even remotely similar to Octoprint in that it just feeds gcode to the firmware. I'm probably talking out of turn as I have only hear Mainsail in passing. (off to do some research now)
> 
> Moonraker (the backend) does not directly parse gcode files like Octoprint. Instead it exposes APIs with which clients (such as Mainsail) can access Klipper's `virtual_sdcard` functionality. Octoprint's gcode parser allows plugins like Exclude Region and Cancel Object to intercept gcode commands and exclude them if they fall within an excluded area. The downside to this that there is additional overhead from the gcode processing. Moonraker has very little overhead, thus it can conceivably run together with Klipper on a Pi Zero.
> 
> Fortunately Klipper's own gcode parser exposes APIs with which we can transform gcode commands. Writing an extras module to exclude an area/object wont be that difficult outside of dealing with retractions. However I still believe that without a UI using these kinds of modules would be an exercise in frustration. Consider a plate 10 copies of an object, where one object fails. A simple list of objects is probably not going to be good enough for a user to know which object they need to cancel.
> 
> I would concede that a full gcode visualizer probably isn't necessary for a usable "cancel object" UI. The object metadata includes bounding coordinates, so a simple UI with selectable "boxes" would probably be good enough for cancel object.

alfrix, 2021-01-13:
The proper Gcode for this implementation would be M486 
https://reprap.org/wiki/G-code#M486:_Cancel_Object

MarcPot, 2021-01-13:
Or just CANCEL_OBJECT since it’s Klipper.

https://reprap.org/wiki/G-code#M486:_Cancel_Object

NameOfTheDragon, 2021-07-11:
This is really a needed feature, because with the best will in the world, sometimes parts do pop off the build plate. With a large build plate, it can be an extremely frustrating experience. I'm posting this comment in the hope that it might bump the issue.

zztopper, 2021-08-13:
That's a very useful feature that is requested by many people in community, so maybe this comment would bump the issue

jdobyns, 2021-09-13:
Not sure why a feature request like this would get closed ...

Do we need to create a new one?

This needs to be a feature in Klipper

ElGranPepo, 2021-09-13:
one more for hope to get this implemented!

kageurufu, 2021-09-13:
In progress, there's a lot of moving parts to get right :)

Czarnyszarf, 2021-11-15:
Any updates ? This feature would make octoprint irrelevant as we have webcam support, a way to access it remotely and in future - cancel objects.

The basic stuff for printer management far from home.

TeddOravec, 2021-12-14:
I am also very much looking for this. Those of us with a large bed who like to print many parts at once could really use this feature! Nothing worse than having to cancel a print where 15/16 parts are printing fine but that one will ruin the rest when its spaghetti gets in the way of the other parts.

robthide37, 2021-12-14:
Pause print at start of layer.  Open slicer then replace without that part
but everything else the same and delete your start code and find same z
height then cancel print and start new file . It's not great but better
then losing the print as long as you don't home again it will be seamless

TeddOravec, 2022-01-23:
> 

You're not wrong, and this is what I and likely most of the others here have been doing to compensate, but then you've got to go through all of that hassle and worry about getting a good first layer again on likely smaller footprints, not an ideal situation. If we could simply say "hey stop printing that one" it would save the hassle.

robthide37, 2022-01-23:
Well a solution for now is pause the print at the end of a layer then reslice the file without that object then edit the file to start at that layer and cancel the print then restart it from that point . Its not ideal but its better than losing the whole print if its a big platter

andrey-kk, 2022-02-20:
Really like Klipper and moved from Octoprint more than a year ago now having two Voron and Mega S but this feature is really needed. I often have issues with adhesion especially after changing from one filament to another. Would save a lot of plastic otherwise it is always a decision to stop print and lose good objects or go on and possibly spoil good ones with leftover plastic from popped up objects. Writing this a looking and my v2.4 printing with PLA and half of the objects popped up in the middle of 350mm2 print. So now full of spaghetti...

Arksine, 2022-02-20:
This is being worked on, you can follow progress at PR #4716.  As previously mentioned, there are several moving parts and its taking some time to make sure that all known issues are resolved before merging.

andrey-kk, 2022-02-20:
Thanks! Will follow.

robthide37, 2022-02-22:
I agree but just making sure you shouldn't be redoing the first layer with
my method ? I assume you just meant resuming on the object in which case
ignore what I said lol